jamesturton/shelly-dimmer-stm32 #2: Shelly Dimmer 1 flickering

daxy01, 2020-11-25:
First of all: Excellent work on adding Shelly Dimmer1/2 support to Tasmota!!!

While running your test image v1.3 (https://github.com/arendst/Tasmota/files/5489877/tasmota_shelly_dimmer_v1.3.bin.gz) I'm experiencing flickering when dimming my lights, even when they are above the 10W. My Dimmer is connected to 3x 4W Calex LEDs. The behaviour can be best described as the LEDs having a too-low refresh rate. The flickering is consistent and happens all the time when dimming the lights.

I've made a recording of the flickering, but it is only noticeable when recording in slow-motion (I guess due to the refresh rate of the camera/iPhone): 
[flickering.mp4.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/5595669/flickering.mp4.zip)

This behaviour is not seen with the stock Shelly firmware.

Please feel free to let me know if any more testing or more information is required.

I have ran the debug image you've provided before and here are the logs:
```
00:00:00 CFG: Loaded from flash at FB, Count 236
00:00:00 QPC: Reset
00:00:00 EXS: Set Power, Power 0x00
00:00:00 Project tasmota Shelly Dimmer Keuken Version 9.0.0.3(shelly-dimmer)-2_7_4_5
00:00:00 SHD Shelly Dimmer Driver v1.4
00:00:00 SHD Starting Tx 1 Rx 3
00:00:00 SNS: Hardware Serial
00:00:00 SHD Sending version command
00:00:00 SHD Shelly Dimmer Co-processor Version v51.3
00:00:00 SHD Updating firmware from v51.3 to v51.4 with 14124 bytes
00:00:02 SHD Sending version command
00:00:02 SHD Loading params: 341,2,0,0,0
00:00:02 SHD Frequency: 8,84 Hz (7210 error points)
00:00:03 SHD Frequency: 50,15 Hz (6488 error points)
00:00:04 SHD Frequency: 50,10 Hz (6450 error points)
00:00:05 SHD Frequency: 50,10 Hz (6434 error points)
00:00:05 RSL: stat/tasmota-keuken/SWITCH1T = {"TRIG":"TOGGLE"}
00:00:05 EXS: Set Power, Power 0x01
00:00:05 RSL: tele/tasmota-keuken/STATE = {"Time":"1970-01-01T00:00:05","Uptime":"0T00:00:04","UptimeSec":4,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":4,"MqttCount":0,"POWER":"ON","Dimmer":12,"Fade":"ON","Speed":1,"LedTable":"OFF"}
00:00:05 RSL: stat/tasmota-keuken/RESULT = {"POWER":"ON"}
00:00:05 RSL: stat/tasmota-keuken/POWER = ON
00:00:06 SHD Frequency: 50,10 Hz (6467 error points)
00:00:06 WIF: Connecting to AP1 Daxy Channel 6 BSSId B6:FB:E4:1B:50:0C in mode 11N as tasmota-keuken-7987...
00:00:07 SHD Frequency: 50,10 Hz (6468 error points)
00:00:07 SHD Frequency: 50,10 Hz (6435 error points)
00:00:07 WIF: Connected
11:39:48 HTP: Web server active on tasmota-keuken-7987 with IP address 192.168.1.113
11:39:48 SHD Frequency: 50,10 Hz (6433 error points)
11:39:49 MQT: Attempting connection...
11:39:49 MQT: Connected
11:39:49 MQT: tele/tasmota-keuken/LWT = Online (retained)
11:39:49 MQT: cmnd/tasmota-keuken/POWER = 
11:39:49 MQT: tele/tasmota-keuken/INFO1 = {"Module":"Shelly Dimmer 1","Version":"9.0.0.3(shelly-dimmer)","FallbackTopic":"cmnd/shelly-dimmer-keuken_fb/","GroupTopic":"cmnd/tasmotas/"}
11:39:49 MQT: tele/tasmota-keuken/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-keuken-7987","IPAddress":"192.168.1.113"}
11:39:49 MQT: tele/tasmota-keuken/INFO3 = {"RestartReason":"Software/System restart"}
11:39:49 MQT: stat/tasmota-keuken/RESULT = {"POWER":"ON"}
11:39:49 MQT: stat/tasmota-keuken/POWER = ON
11:39:49 SHD Frequency: 50,10 Hz (6430 error points)
11:39:50 MQT: homeassistant/sensor/D0DF33_ANALOG_Temperature/config = {"name":"Shelly Dimmer Keuken ANALOG Temperature","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ANALOG_Temperature","dev":{"ids":["D0DF33"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['ANALOG']['Temperature']}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_ENERGY_TotalStartTime/config = {"name":"Shelly Dimmer Keuken ENERGY TotalStartTime","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ENERGY_TotalStartTime","dev":{"ids":["D0DF33"]},"unit_of_meas":" ","ic":"mdi:progress-clock","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['TotalStartTime']}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_ENERGY_Total/config = {"name":"Shelly Dimmer Keuken ENERGY Total","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ENERGY_Total","dev":{"ids":["D0DF33"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Total']}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_ENERGY_Yesterday/config = {"name":"Shelly Dimmer Keuken ENERGY Yesterday","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ENERGY_Yesterday","dev":{"ids":["D0DF33"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Yesterday']}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_ENERGY_Today/config = {"name":"Shelly Dimmer Keuken ENERGY Today","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ENERGY_Today","dev":{"ids":["D0DF33"]},"unit_of_meas":"kWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Today']}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_ENERGY_Power/config = {"name":"Shelly Dimmer Keuken ENERGY Power","stat_t":"tele/tasmota-keuken/SENSOR","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D0DF33_ENERGY_Power","dev":{"ids":["D0DF33"]},"unit_of_meas":"W","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Power']}}"} (retained)
11:39:50 MQT: homeassistant/light/D0DF33_LI_1/config = {"name":"Keuken schakelaar","stat_t":"tele/tasmota-keuken/STATE","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/tasmota-keuken/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D0DF33_LI_1","dev":{"ids":["D0DF33"]},"bri_cmd_t":"cmnd/tasmota-keuken/Dimmer","bri_stat_t":"tele/tasmota-keuken/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}"} (retained)
11:39:50 MQT: homeassistant/sensor/D0DF33_status/config = {"name":"Shelly Dimmer Keuken status","stat_t":"tele/tasmota-keuken/HASS_STATE","avty_t":"tele/tasmota-keuken/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota-keuken/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"D0DF33_status","dev":{"ids":["D0DF33"],"name":"Shelly Dimmer Keuken","mdl":"Shelly Dimmer 1","sw":"9.0.0.3(shelly-dimmer)","mf":"Tasmota"}} (retained)
11:39:50 SHD Frequency: 50,10 Hz (6441 error points)
11:39:51 SHD Frequency: 50,10 Hz (6442 error points)
11:39:52 SHD Frequency: 50,10 Hz (6448 error points)
11:39:53 MQT: tele/tasmota-keuken/STATE = {"Time":"2020-11-08T11:39:53","Uptime":"0T00:00:13","UptimeSec":13,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":97,"MqttCount":1,"POWER":"ON","Dimmer":12,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Daxy","BSSId":"B6:FB:E4:1B:50:0C","Channel":6,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:39:53 MQT: tele/tasmota-keuken/SENSOR = {"Time":"2020-11-08T11:39:53","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":45.4},"ENERGY":{"TotalStartTime":"2020-11-06T15:36:51","Total":5955.004,"Yesterday":0.000,"Today":2976.304,"Period":0,"Power":3},"TempUnit":"C"}
11:39:53 SHD Frequency: 50,10 Hz (6475 error points)
11:39:54 SHD Frequency: 50,10 Hz (6484 error points)
11:39:55 SHD Frequency: 50,10 Hz (6475 error points)
11:39:56 SHD Frequency: 50,10 Hz (6459 error points)
11:39:57 SHD Frequency: 50,5 Hz (6468 error points)
11:39:58 SHD Frequency: 50,15 Hz (6449 error points)
[... 47 lines cut ...]
11:40:30 MQT: tele/tasmota-keuken/STATE = {"Time":"2020-11-08T11:40:30","Uptime":"0T00:00:50","UptimeSec":50,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":39,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Daxy","BSSId":"B6:FB:E4:1B:50:0C","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:40:30 MQT: stat/tasmota-keuken/RESULT = {"POWER":"ON","Dimmer":50}
11:40:30 SHD Frequency: 50,5 Hz (6440 error points)
11:40:31 SHD Frequency: 50,15 Hz (6455 error points)
[... cut ...]
11:40:43 MQT: tele/tasmota-keuken/STATE = {"Time":"2020-11-08T11:40:43","Uptime":"0T00:01:03","UptimeSec":63,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":83,"MqttCount":1,"POWER":"OFF","Dimmer":38,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Daxy","BSSId":"B6:FB:E4:1B:50:0C","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:40:43 MQT: stat/tasmota-keuken/RESULT = {"POWER":"OFF"}
11:40:43 MQT: stat/tasmota-keuken/POWER = OFF
11:40:43 SHD Frequency: 50,15 Hz (6424 error points)
11:40:44 SHD Frequency: 50,5 Hz (6412 error points)
11:40:45 SHD Frequency: 50,15 Hz (6423 error points)
```

jamesturton, 2020-11-25:
@daxy01 Thanks for taking the time to test this firmware!
After scavenging my house and buying some more bulbs to test with I have managed to find one bulb which this 'flickering' effect is just about visible, so I should now be set up for doing some more experimenting and testing.
Will keep you updated with my findings!

digitalface, 2020-11-26:
I have mine in another room ready to test if you update the f/w etc. Not sure how else I can help really.

jamesturton, 2020-11-27:
@digitalface @daxy01 I've looked over the code and I have identified two things that might perhaps be causing the flickering.

Firstly, I saw there was no pull-up/pull-down set on the mains detect pin - this meant that it was possibly susceptible to interference (there might have been hardware pull-up/pull-down but I can't remember). I have enabled an internal pull-down to this pin to avoid this.

Secondly, the value for the mains frequency was being updated twice per cycle, once on the positive half of the cycle and once on the negative part of the cycle. My guess is that the reference (neutral) value that the Shelly dimmer uses is slightly distorted, thus one half of the cycle is longer than the other. This would cause the calculated brightness value to fluctuate every half cycle too! I have tried to fix this by averaging the line frequency over a whole cycle.

You can find the changes I have made in the branch [fix-flicker](/jamesturton/shelly-dimmer-stm32/tree/fix-flicker).

And the binary for you to test with is here: [shelly-dimmer-stm32_v51.5.bin.gz](https://github.com/jamesturton/shelly-dimmer-stm32/files/5608974/shelly-dimmer-stm32_v51.5.bin.gz)

You can use the [latest development build of Tasmota](http://ota.tasmota.com/tasmota/) to upload the firmware to the stm32 directly using the firmware upgrade page. Just unzip the binary and select it in the "Upgrade by file upload" input.

Let me know if this makes any difference!

daxy01, 2020-11-27:
Hi @jamesturton. Thanks for your help on this!
It seems I can't run this. I've upgraded to the latest dev build (9.1.0.2). However, I can't control the Shelly anymore, the dim slider is not shown. I can't upgrade stm32 either, it tells me "Upload Failed - Invalid file signature".

Quickly checking, the template didn't change did it? This is what I used:
`{"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5504,5536,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}`

I've now downgraded back to the v1.3 version you've created earlier. I'm not quite sure what is happening and why it fails :(

digitalface, 2020-11-27:
Hi @jamesturton , 

I must be doing something wrong - I've updated to 9.1.0.2 and tried to upload the .bin file but get "invalid signature" - I then uploaded the .gz file which bricked my Shelly, Fortunately I flashed Tasmota back using USB - TTL tool on the pin headers.

Your advise is appreciated on this 👍

jamesturton, 2020-11-28:
@daxy01 
> Quickly checking, the template didn't change did it? This is what I used:
> `{"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5504,5536,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}`

It looks like the template has changed slightly, I'm not too sure why. I guess when arendst merged the code into the main Tasmota repository something changed. You can find the working template on the blakadder template site: [Shelly dimmer template](https://templates.blakadder.com/shelly_Dimmer.html).

@digitalface maybe you could also try updating your template too, and then try uploading the .bin file again?

daxy01, 2020-11-28:
That did the trick!
I've now been able to upgrade to the latest dev version (and still be able to control it!) and applied the stm upgrade.
Initial feedback is that it does improve significantly but when dimming at roughly 8W (which is less than the 10W minimum), minimal flickering is still observed. Still, this seems a lot better so far!!!

Again, this is feedback based on 2 min looking at the lamps 😀 I will perform further testing this weekend.

Slightly off-topic: I will mention the Tamplate change in Tasmota Issues for both the Dimmer 1 and 2 topics, just to make sure everyone who's on your test image knows this has changed.

Thanks again for your work!!!

digitalface, 2020-11-28:
@jamesturton Yeah that's worked now, updated the template then uploaded the .bin file. Difficult to see if it's made a difference yet, will have to wait until night time to see :) Will update the thread later.

ghuening, 2020-11-28:
Hi all, I need help.
I'm on a shelly dimmer 2, have flashed it with Tasmota 9.1.0.2 successfully. On it's webpage I can see 
tempature, and power (1-2W without load). I can switch on/off and move slider. If I connect a bulb (normal 40W,230V) nothing happens, the light stays dark. My template is 
{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}

With original firmware it was working well, same schematic, N connected. What is wrong?

best regards
Guenter

daxy01, 2020-11-28:
> Hi all, I need help.
> I'm on a shelly dimmer 2, have flashed it with Tasmota 9.1.0.2 successfully. On it's webpage I can see
> tempature, and power (1-2W without load). I can switch on/off and move [...]

Try the official template as per: https://templates.blakadder.com/shelly_Dimmer_2.html
`{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}`
It's slightly different than yours.

ghuening, 2020-11-28:
@daxy01 
Thanks for anwser, but this also don't work. Web interface seems to be ok, but no output :-(.

daxy01, 2020-11-28:
> @daxy01
> Thanks for anwser, but this also don't work. Web interface seems to be ok, but no output :-(.

Perhaps your bulb is defective? 😀
As your issue doesn't seem to be STM32 related but rather a generic Tasmota issue, I would suggest opening a Tasmota issue at https://github.com/arendst/Tasmota/issues/new

JameZUK, 2020-11-28:
@daxy01 I had the exact same problem. Seemed to be working but no output. Eventually I managed to get it working by flashing the stm firmware again. 
These steps seem to work well:
1. Flash latest tasmota Dev version 9.1.0.2
2. Apply correct template from here https://templates.blakadder.com/shelly_Dimmer_2.html
3. Flash the stm firmware using the week GUI. 

Check the version number of the STM firmware in the console just after restarting, if it says v0.0 you need to try flashing the STM again. 

This worked for me anyway so it hope it helps someone!

@jamesturton nice work on this mate, very cool in fact. Thanks

digitalface, 2020-11-28:
Okay so it's dark now and I've been able to test my 3x GU10 5w bulbs. They're no longer flickering which is great.  They do buzz more than with original f/w but only at low (10-20%) setting. The power measurement is also working perfectly where it didn't before, but maybe that's to do with the Tasmota software level?

Great work @jamesturton. I was getting so fed up of the delay with the control responses using the original Shelly frmware. Tasmota fixes that.

ghuening, 2020-11-28:
@JameZUK 
Hi James,
I have no idea how to flash stm firmware and how can I check the version? Is there a description somewhere?
BR

digitalface, 2020-11-28:
@ghuening If you restart the dimmer and log back in, check the console page and you'll see something like "Shelly Dimmer Co-processor Version " . You're expecting to see v51.5

rbswift, 2020-11-29:
Thanks for persisting with this!

I have updated but still getting quite bad flickering when dimming 4 x [9W SAL downlights](https://www.sal.net.au/products/interior-led-lighting/led-downlights/wave-s9065tc). Is there anything obvious wrong below or any suggestions?

`00:00:00 CFG: Loaded from flash at F9, Count 277
00:00:00 QPC: Count 1
00:00:00 Project tasmota Dining Version 9.1.0.2(tasmota)-2_7_4_7
00:00:00 SNS: Hardware Serial
00:00:00 SHD: Shelly Dimmer Co-processor Version v51.3
00:00:00 WIF: Connecting to AP1 Teh Interwebz Channel 6 BSSId 08:96:D7:8A:94:84 in mode 11N as dining...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on dining with IP address 192.168.188.33
04:41:34 MQT: Attempting connection...
04:41:34 MQT: Connected
04:41:34 MQT: tele/dining/LWT = Online (retained)
04:41:34 MQT: cmnd/dining/POWER = 
04:41:34 MQT: tele/dining/INFO1 = {"Module":"Shelly Dimmer 1","Version":"9.1.0.2(tasmota)","FallbackTopic":"cmnd/dining_fb/","GroupTopic":"cmnd/tasmotas/"}
04:41:34 MQT: tele/dining/INFO2 = {"WebServerMode":"Admin","Hostname":"dining","IPAddress":"192.168.188.33"}
04:41:34 MQT: tele/dining/INFO3 = {"RestartReason":"Software/System restart"}
04:41:34 MQT: stat/dining/RESULT = {"POWER":"ON"}
04:41:34 MQT: stat/dining/POWER = ON
04:41:37 QPC: Reset
04:41:38 MQT: tele/dining/STATE = {"Time":"2020-11-29T04:41:38","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":54,"MqttCount":1,"POWER":"ON","Dimmer":22,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Teh Interwebz","BSSId":"08:96:D7:8A:94:84","Channel":6,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
04:41:38 MQT: tele/dining/SENSOR = {"Time":"2020-11-29T04:41:38","2,32000,10000,33500000,0":"OFF","Switch2":"OFF","ANALOG":{"Temperature":64.5},"ENERGY":{"TotalStartTime":"2020-11-05T11:23:12","Total":0.001,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0},"TempUnit":"C"}
04:41:50 MQT: stat/dining/RESULT = {"POWER":"OFF"}
04:41:50 MQT: stat/dining/POWER = OFF
04:41:51 MQT: stat/dining/RESULT = {"POWER":"ON"}
04:41:51 MQT: stat/dining/POWER = ON
04:41:52 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":37}
04:41:54 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":30}
04:42:07 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":58}
04:42:48 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":100}
04:42:58 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":34}
04:43:08 MQT: stat/dining/RESULT = {"POWER":"ON","Dimmer":100}`

digitalface, 2020-11-29:
Looks like you're in the previous STM firmware 1.3, you need to update to the latest.

rbswift, 2020-11-29:
Thanks @digitalface I've got v51.5 now and it definitely seems a bit better. Still a bit of flicker I think but less noticeable. I will try to do a side by side comparison before upgrading the rest of my dimmers. Cheers

rbswift, 2020-11-29:
On this page: https://tasmota.github.io/docs/Commands/
it states that the dimmerrange command "Does not change Dimmer command behavior" is that still correct? It seems like the dimmer command 0-100 takes into account the dimmerrange. I actually prefer this behaviour but just want to check I'm not doing something wrong?

rbswift, 2020-11-29:
also are ShdWarmupBrightness and ShdWarmupTime executed at restart? I seem to have trouble getting a minimum brightness initially after restart

rbswift, 2020-11-29:
is LoadAvg:99 normal? even when dimmer is off?

apologies - just realised this is probably not the best place for general feedback. where's better?

ghuening, 2020-11-29:
@digitalface , @JameZUK , @daxy01 
stm update  done and now it is working fine, great support !!!
Thx all for your help

@jamesturton thx for your work, well done!

BR
Guenter

jamesturton, 2020-11-30:
@JameZUK Thanks for writing up the guide to flashing this firmware using the development version of Tasmota! It seems like you have really help some people out!

@rbswift 
> On this page: https://tasmota.github.io/docs/Commands/
> it states that the dimmerrange command "Does not change Dimmer command behavior" is that still correct? It seems like the dimmer command 0-100 [...]

It does change the dimmer behaviour using this driver. For example using the command `dimmerrange 0 50` would mean Tasmota would divide all `dimmer` values by 2 before sending it to the stm32. Similarly `dimmerrange 50 100` would be the same as dividing by 2 and then addind 50 to the value.

> also are ShdWarmupBrightness and ShdWarmupTime executed at restart? I seem to have trouble getting a minimum brightness initially after restart

`ShdWarmupBrightness` and `ShdWarmupTime` aren't implemented in this firmware. The commands still exist in Tasmota from use with the Shelly firmware but I think these commands are already implemented in Tasmota as [`Wakeup`](https://tasmota.github.io/docs/Commands/#wakeup) and [`WakeupDuration`](https://tasmota.github.io/docs/Commands/#wakeupduration) so maybe I should remove `ShdWarmupBrightness` and `ShdWarmupTime` from the driver.

rbswift, 2020-11-30:
@jamesturton 
Thanks for the confirmation re dimmerrange - that's great.
I think wakeup is a bit different to warmup. Wakeup is intended to gently and gradually increase a light's brightness in the morning taking up to 50 minutes. Warmup is more of a kickstart when switching on, for example from 0/off to 5% dim doesn't provide enough power to fire up to lights so they are very briefly lit to a higher brightness for a fraction of a second and then dimmed to the target level. I do kind of agree this might be more useful as a universal tasmota feature than in a specific driver. It could possibly be inelegantly achieved with backlog I guess?

jamesturton, 2020-11-30:
@rbswift I see, well maybe a `wakeup` command should be implemented in Tasmota, but I also agree that it would be best for it to be a universal feature - not just for the Shelly dimmer driver. 
Unfortunetally I have no idea about `LoadAvg`. Mine report as ~20 if that helps at all. If you think it's a bug I would recommend opening an issue on the [Tasmota repo](https://github.com/arendst/Tasmota/issues).

daxy01, 2020-11-30:
I've been testing v51.5 for the past days and I would say it works fine.
As mentioned, when I dim to <40% (with DimmerRange being 25,100) I can see a bit of flickering, but this is significantly less than before (e.g. doesn't bother me much 😀). I guess this is due to the low load (4-7W when dimmed) so I'm not concerned.
Anything above 40% it works perfectly for me! 

One thing I noticed is that my Shelly reports a temp of -275 C. Does anyone else have this as well or is it just me?

@jamesturton, I believe Tasmota (dev) will be shipped with stm32 v51.4 if I'm correct. May I suggest to have v51.5 included before the official release of the new Tasmota (I believe that's 9.1.0.1 isn't it).

jamesturton, 2020-11-30:
Great! I shall close this issue make a release!
@daxy01 the temperature is taken from the ADC pin which was renamed to GPIO17 in Tasmota 9.1. Can you verify that GPIO 17 is set to `ADC Temp` in the configure template page? 
[image]

closed 2020-11-30 by jamesturton

daxy01, 2020-11-30:
@jamesturton Yes, my settings are exactly the same.
I even tried setting GPIO17 to None and then back to ADC Temp, but it remains around -275 C. I just did a hard reset and it doesn't change. It's a minor thing, I'm not bothered if its wrong just curious.

digitalface, 2020-11-30:
@daxy01 mine appears to be reporting the correct temperature, always between 38c and 41c depending on dimmer setting.

arendst, 2020-12-02:
Pls compare your @digitalface and @daxy01 ``adcparam`` parameters.

arendst, 2020-12-02:
@jamesturton do you see a possibility to PR your latest stable stm32 code into tasmota. I would like to have a stable copy in my repository. (Its in folder ``/tools/fs_ShellyDimmer_stm32``)

daxy01, 2020-12-02:
@digitalface 
As recommended by @arendst (Thanks!), this is my output:
```
13:54:25 CMD: adcparam
13:54:25 MQT: stat/tasmota-keuken/RESULT = {"AdcParam1":[2,32000,10000,1.1104]}
```

jamesturton, 2020-12-02:
@daxy01 my `adcparam` looks like this if you want to try:
```
{"AdcParam1":[2,32000,10000,3350]}
```

digitalface, 2020-12-02:
14:02:04 CMD: adcparam
14:02:04 RSL: RESULT = {"AdcParam1":[2,32000,10000,3350]}

daxy01, 2020-12-02:
> 2,32000,10000,3350
@jamesturton @digitalface 
Awesome! That's a lot better (and default settings). Thanks, and thanks to @arendst for the hint.
Weird that it changed, for sure I didn't touch it 😀.

JessePrater, 2021-01-26:
Just wanted to say thanks to you guys for working on this- I can report as well that the one light (chandelier with 7 Osram bulbs) I had a flickering issue with is MUCH better now. (Especially after disabling leading edge again- forgot I tried that while troubleshooting.)

Interestingly, all needed the bottom dimmerrange reset to ~10-15 to avoid strobing. Even on the Cree lights that worked fine before.

PR #3 (2020-11-30): Fix flicker
Closes #2