jamesturton/shelly-dimmer-stm32 #5: Invalid file signature

JohnnyPicnic, 2020-12-06:
I get an "Invalid file signature" when trying to upload the shelly-dimmer-stm32_v51.5.bin file.
Running Tasmota 9.1.0.2

rbswift, 2020-12-06:
Did you set and activate the template first? I think it might not work until you do that.

https://templates.blakadder.com/shelly_Dimmer.html

JohnnyPicnic, 2020-12-06:
Yup that did it. Thanks!

closed 2020-12-06 by jamesturton

laagland, 2020-12-12:
I'm having the same problem.
It's a Shelly Dimmer 2 with Tasmota 9.1.0.2 and I have activated the correct template from the blakadder site. Still, everytime I try to upload the firmware, it says "Invalid file signature"

Any idea would could be wrong?

rbswift, 2020-12-13:
@laagland Maybe post your full console log from after the error along with the template you are using and the URL for the firmware file you are trying to upload so we can check for you.

laagland, 2020-12-13:
@rbswift The console doesn't log much:
`13:56:31 UPL: File shelly-dimmer-stm32_v51.5.bin ...`

That's it.

I've used the template from this page: [https://templates.blakadder.com/shelly_Dimmer_2.html](https://templates.blakadder.com/shelly_Dimmer_2.html) and tries to upload the bin file linked on the same page.

laagland, 2020-12-13:
Ok nevermind. I had only flashed tasmota-lite when trying. After upgrade to full with OTA and a powercycle I was able to flash the stm.

Thanks anyway!