jamesturton/shelly-dimmer-stm32 #6: Shelly Dimmer 2 - Flickering and Brightness issues

luke3butler, 2020-12-13:
First I want to thank you for the work you've put into adding support for these dimmers in Tasmota.

I'm somewhat new to using Tasmota, but I'm competent with hardware/software, so if there's anything you need me to try I'm all for it.

Tried on two devices (different set of lights as well) with the same results.

## Brightness Level
At (or less than) 46% the lights are fully off, 47% the lights look like they're at 1%. 
46%-100% is pretty much a normal 0-100% linear brightness scale.

## Flickering
It's quite bad, and I'll have to keep this device on full brightness until it's fixed. The flicker is more noticeable the lower the brightness gets, but even at higher levels it's noticeable.

I noticed that you fixed some flickering issues in the latest release, so I was a bit disappointed that I ran into this one.

## Steps I took

- Flashed Tasmota 9.1.0.2
- Ran `reset 5` in the console
- Applied the template from [here](https://templates.blakadder.com/shelly_Dimmer_2.html)
- Flashed the firmware from the releases page on this repo (v51.5)
- Several reboots afterwards

rbswift, 2020-12-13:
**Brightness Level** - try these 2 commands
LedTable 0
DimmerRange 20,100

**Flickering** - check the Console log and make sure you actually see version 51.5, if not then upgrade again
00:00:00 SHD: Shelly Dimmer Co-processor Version v51.5

luke3butler, 2020-12-13:
Thanks for the quick response!

**Brightness Level**
`LedTable 0` takes me back to the main menu, but I did run it 3 times.
`DimmerRange 20,100` no change whatsoever
`DimmerRange 40,100` no change whatsoever.
Seems like it's not really doing anything? Rebooted a few times to be sure.

**Flickering**
Re-Flashed just to be sure

> 00:00:00 Project tasmota Kitchen Ceiling Lights Version 9.1.0.2(tasmota)-2_7_4_9(2020-12-12T18:32:47)
> 00:00:00 SNS: Hardware Serial
> 00:00:00 SHD: Shelly Dimmer Co-processor Version v51.5

Still flickering unfortunately.

rbswift, 2020-12-13:
maybe just try `ledtable` on its own and check that it's off:

```
03:09:33 CMD: ledtable
03:09:33 MQT: stat/kitchen/RESULT = {"LedTable":"OFF"}
```

luke3butler, 2020-12-13:
```
LedTable 0
DimmerRange 20,100
```
Fixed the brightness issue after all. 
Thanks for your help on that one. Seems like something that should be default for this device?

Just down to this flickering issue now.

rbswift, 2020-12-13:
unsure why you've got 2_7_4_9 when I've got 2_7_4_7 but I doubt that's really relevant.
```
00:00:00 Project tasmota Dining Version 9.1.0.2(tasmota)-2_7_4_7
00:00:00 SNS: Hardware Serial
00:00:00 SHD: Shelly Dimmer Co-processor Version v51.5
```
You could check the `ShdLeadingEdge` - should be 0 for trailing edge dimming and 1 for leading edge.

luke3butler, 2020-12-13:
> 05:32:38 MQT: stat/tasmota_00E800/RESULT = {"ShdLeadingEdge":0}

I went ahead and switched it with `ShdLeadingEdge 1`.

Still flickers a slight bit on lower brightness, but I think it isn't noticeable enough at most brightness options for it to matter. I've also been staring at these flickering lights for a while and will need to give my eyes a break to know for sure.

jamesturton, 2020-12-13:
@luke3butler Do you use 50Hz or 60Hz mains in your region? I haven't been able to test this firmware with 60Hz so there could be a problem there...?
Which kind of bulb are you trying to power?

jamesturton, 2020-12-19:
I take it these settings helped out? If you are still having problems let me know and I will re-open this issue.

luke3butler, 2020-12-21:
Apologies on the late response.

I am on 60Hz electricity, yes. I've tried a few different LED bulbs, including some Amazon Basics dimmable bulbs. No apparent flicker when using stock Shelly firmware.

There's still flickering after these changes, although not quite as badly.

Flicker is most noticable under 30% brightness.

jamesturton, 2020-12-22:
Ok, reopening this issue as it doesn't sound resolved.
Finding the problem might be a little hard if I can't replicate the issue here, but I'll see what I can find!

jamesturton, 2020-12-22:
@luke3butler just to confirm, are you using a neutral wire in your setup?

luke3butler, 2020-12-23:
@jamesturton yes.

jamesturton, 2021-01-04:
Hi again. I have made two test versions for you. Please can you try these out and let me know if either of them help with the problem? The version 'beta-2' might act a little different, but it would be good if you could tell me if it helps at all with the flicker.
[shelly-dimmer-stm32_v51.5_betas.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/5765542/shelly-dimmer-stm32_v51.5_betas.zip)

luke3butler, 2021-01-06:
@jamesturton So I flashed between the original 51.5, beta1, and beta2 three times so I could be sure of the differences.

**Beta 1**
Better than the original. Still flickers a bit, but a noticeable improvement.

**Beta 2**
It gave me a lot more range in the lower end (didn't know these lights could dim any lower).
Flicker can be seen very clearly on the lower end, but when at the equivalent of 5-10% brightness of the original firmware (and beta 1) the flicker is almost entirely gone. My wife says it's not noticeable enough to matter, so it passes the wife test there.

If I adjusted the DimmerRange for beta 2 to compensate, it'd be good enough for me.

Thanks for the time you've put into this. If you have any more builds for me to test, need any more info, or want me to try getting a video, just let me know.

Dilergore, 2021-01-16:
Hey,

I have a similar issue:
https://github.com/arendst/Tasmota/issues/9412#issuecomment-761508706

I am also happy to test if you need.

jamesturton, 2021-04-22:
Hi all,

I have a new version to test, which (in my testing) has reduced the flickering when used in a no-neutral setup. I hope this also improves the flickering in a normal neutral setup too.
One thing to note is that it seems sometimes the leading edge property isn't sent correctly from Tasmota, so I can recomend sending `ShdLeadingEdge 0` command after updating the firmware.

Please let me know your feedback!

[shelly-dimmer-stm32.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/6360141/shelly-dimmer-stm32.zip)

James

Jas0r, 2021-05-02:
Hi
I have a set of lights in my living that have been flickering using the Shelly Dimmer 2 using the Shelly firmware. I have tried all versions up to the latest. I have moved to tasmota and applied your latest patch and run it for a week now. I have had no flickers or issues. 
My lights running are 4 of the below:
https://limelighting.com.au/ziiluxe_products/origin-modular-13/

jamesturton, 2021-05-02:
@Jas0r and @rnauber thanks for testing this version. As you both had positive experiences with this version I will release an official version and mark this issue and #7 as fixed!

jj-uk, 2021-08-31:
I'm still seeing some flickering here. Running 51.6.

Ledtable is off, dimmer range is 10,100. Leading edge is 1.

Setting the bulb to 40%, the dimming looks fine if looking directly at the bulb, but if you look away so the bulb is in your peripheral vision, the flickering is very clear to see.

Lower brightness settings show the issue more prominently. 

I'm on 50Hz.

jamesturton, 2021-09-07:
Hi @jj-uk, you are using dimmable bulbs right?

I assume you are using Tasmota on the ESP?
Have you tried using the `warmup_brightness` setting? You can choose a value between 10 and 100. See if that helps

jj-uk, 2021-09-19:
I've bought new bulbs to test with. They are Philips bulbs.  They still flicker at between 50 and 80%.

Tasmota is rejecting "warmup_brightness" as a command in the console.

rbswift, 2021-09-19:
`ShdWarmupBrightness`, `ShdWarmupTime`, and `ShdLeadingEdge` are the 3 shelly dimmer specific commands I believe. Maybe note the model number of the bulbs you are using and how many in parallel.

jj-uk, 2021-09-19:
OK thanks. I'll try the command at a later time.

The bulbs are "Philips Master 6W LEDluster E14 SES Golf Ball DimTone" - https://www.ledbulbs.co.uk/products/philips-6w-led-e14-ses-golf-ball-very-warm-white-dimmable-45358200

I have 5 in parallel.

lngr, 2021-10-03:
Hi,

I have the same flickering issues at 100% brightness and a classic halogene light bulb.

```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FB, Count 18
00:00:00.055 QPC: Count 1
00:00:00.065 Project tasmota shdm2-buero-marie-wand Version 9.5.0(tasmota)-2_7_4_9(2021-06-17T08:26:35)
00:00:00.066 SNS: Hardware Serial
00:00:00.170 SHD: Shelly Dimmer Co-processor Version v51.5
```

The beta2 above removes the flickering.  The flickering is still there with the 51.6 release. Not sure what's the difference between 51.6 and the beta2 from above, but the beta2 works well for me.

kratz00, 2021-10-05:
Hi

I also have flickering in a similar setup like @lngr does.
I went back to v51.5 beta 2 and the flickering is gone for me too, the light also switches on/off much faster now.
So some change(s) from v51.5 beta 2 to v51.6 seems to have introduced this problem.

jj-uk, 2021-10-05:
Is this what you guys are running?
https://github.com/jamesturton/shelly-dimmer-stm32/releases/tag/v51.5

rbswift, 2021-10-05:
> … the light also switches on/off much faster now.

@kratz00 Do you mean when using an external switch via the SW1 or SW2 connections? I find Tasmota sometimes takes several seconds to sense the switch which is really frustrating and would love a way to fix that!

jj-uk, 2021-10-05:
Ah ok. I'm not that familiar with github. I can only find "Releases". What needs clicking to get to archives?

lngr, 2021-10-06:
That's the one I'm using, too.

jamesturton, 2021-10-07:
@lngr Just to confirm, you are using the Shelly dimmer 2?

lngr, 2021-10-07:
@jamesturton yes

jamesturton, 2021-11-07:
> > … the light also switches on/off much faster now.
> 
> @kratz00 Do you mean when using an external switch via the SW1 or SW2 connections? I find Tasmota sometimes takes several seconds to sense the switch which is really frustrating and would love a way to fix that!

@rbswift what settings do you have in Tasmota for `SwitchDebounce` and `SetOption32`?

rbswift, 2021-11-07:
> @rbswift what settings do you have in Tasmota for `SwitchDebounce` and `SetOption32`?

SwitchDebounce 100. I was getting phantom switching at default 50
SetOption32 40 (default)

thanks for any guidance. I’m using a rocker switch, not a momentary button btw

elibl, 2021-11-13:
Hello together,
I've got also some very disturbing stroboscopic effect in let's say 50% or even more use time. Sometimes it's totally stable. The occurence looks to be random so can't really tell what are the circumstances.

My setup:
230V / 50Hz
No neutral with Shelly Bypass
Osram dimmable led bulbs

`00:00:00.001 HDW: ESP8266EX
00:00:00.051 CFG: aus Flash geladen am F4, Anzahl 879
00:00:00.056 QPC: Count 1
00:00:00.067 Projekt tasmota Decke Version 9.5.0(tasmota)-2_7_4_9(2021.06.17 08:28:30)
00:00:00.068 SNS: Hardware Serial
00:00:00.171 SHD: Shelly Dimmer Co-processor Version v51.6
00:00:00.775 WIF: verbinden mit AP1 ****** Channel 6 BSSId **:**:**:**:**:** in Modus 11n als tasmota_******-6150...
00:00:04.753 WIF: verbunden
00:00:05.011 HTP: Web-Server aktiv bei tasmota_******-6150 mit IP-Adresse ***.***.***.***
10:04:04.679 MQT: Verbindungsversuch...
10:04:04.697 MQT: verbunden
10:04:04.700 MQT: tele/tasmota_******/LWT = Online (beibehalten)
10:04:04.702 MQT: cmnd/tasmota_******/POWER = 
10:04:04.712 MQT: tele/tasmota_******/INFO1 = {"Info1":{"Module":"Shelly Dimmer 2","Version":"9.5.0(tasmota)","FallbackTopic":"cmnd/DVES_******_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:04:04.723 MQT: tele/tasmota_******/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_******-6150","IPAddress":"***.***.***.***"}}
10:04:04.734 MQT: tele/tasmota_******/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
10:04:04.744 MQT: stat/tasmota_******/RESULT = {"POWER":"ON"}
10:04:04.748 MQT: stat/tasmota_******/POWER = ON
10:04:04.781 QPC: Reset
10:04:05.738 MQT: stat/tasmota_******/RESULT = {"Time":"2021-11-13T10:04:05","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":179,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"******","BSSId":"**:**:**:**:**:**","Channel":1,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:04:05.791 MQT: stat/tasmota_******/STATUS10 = {"StatusSNS":{"Time":"2021-11-13T10:04:05","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":42.3},"ENERGY":{"TotalStartTime":"2021-07-23T08:54:46","Total":41126.801,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"TempUnit":"C"}}
10:04:08.763 MQT: tele/tasmota_******/STATE = {"Time":"2021-11-13T10:04:08","Uptime":"0T00:00:14","UptimeSec":14,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"******","BSSId":"","Channel":1,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:04:08.775 MQT: tele/tasmota_******/SENSOR = {"Time":"2021-11-13T10:04:08","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":42.3},"ENERGY":{"TotalStartTime":"2021-07-23T08:54:46","Total":41126.801,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":13,"ApparentPower":13,"ReactivePower":0,"Factor":1.00,"Voltage":226,"Current":0.059},"TempUnit":"C"}`

If there are further logs I can contribute just tell me the commands to get them.

Thanks a lot for the awesome work so far.

sparkplug23, 2021-12-01:
I am wondering if this is the same issue. I am seeing every 1 second, the light dims maybe 20% from set brightness. My first time using the latest firmware, as my other shelly dimmer 2 have successfully been using the version released last year. The bulb is a normal halogen style, so no issue with leds. I will try flashing the older firmware tomorrow to see if it works, but I was just curious if this is the type of flickering discussed here.

megakid, 2021-12-11:
I'm also getting 51.6 flickering - 51.5 is fine.  240V 50Hz in UK

sparkplug23, 2021-12-11:
> I'm also getting 51.6 flickering - 51.5 is fine.  240V 50Hz in UK

I went back to 51.5 for now, but I'd love to use the no neutral on one device so I need 51.6 for that. When I've more time I'll do further testing.

Ara-Nova, 2022-12-14:
unfortunately my dimmer only works, when im using . ShdLeadingEdge 1 and then ShdLeadingEdge 0.

after a few switch tapping my Shelly Dimmer 2 isnt working anymore. i have to reenter ShdLeadingEdge 1 and 0 . The config will not be saved. savedata0 and savedata1 not working.

Please help

a