jamesturton/shelly-dimmer-stm32 #8: Shelly Dimmer 2 Module  Tasmota  Upload Failed  Invalid file signature

jannodeluxe, 2020-12-14:
I get the following error when trying to flash the 51.5 FW:

Shelly Dimmer 2 Module
Tasmota
Upload Failed
Invalid file signature

Upgraded to 9.1.0.2
console reset 5
then tried to flash the bin file

jamesturton, 2020-12-14:
@jannodeluxe Have you followed the instructions here: https://templates.blakadder.com/shelly_Dimmer_2.html
You must set the template before uploading new firmware.

jannodeluxe, 2020-12-14:
hi @jamesturton thanks for your fast reply. Yes I did absolutely that way.

flashed Tasmota 9.1.0.2 lite
applied template
reset 5
then tried to flash the 51.5 bin file

jamesturton, 2020-12-14:
> flashed Tasmota 9.1.0.2 lite
> reset 5
> applied template
> then tried to flash the 51.5 bin file

The Shelly dimmer drive is not included in in the Tasmota 'lite' version. You must use the full [`tasmota.bin.gz` version](http://ota.tasmota.com/tasmota/). Please try updating to that version and set the template again.

jannodeluxe, 2020-12-14:
That made the trick! Thanks in advanced!

closed 2020-12-14 by jannodeluxe