jamesturton/shelly-dimmer-stm32 #9: Tasmota Template Difference

mgoeller, 2020-12-15:
Hi

Thank you for the open source development of your STM firmware. I am interested to use this firmware with my own ESP8266 firmware, and the serial protocol is understandable and works fine.

However only, when I flash the tasmota_shelly_dimmer_v1.3.bin.gz firmware file from the [support thread](https://github.com/arendst/Tasmota/issues/6914) first.

If I update the STM firmware to v51.5, I cannot control the STM anymore, and I guess there is a difference in initialisation, right?

Would you be so kind and tell me how to initialise GPIO 4 and 5? I saw that the template is different from the combined binary to the official Tasmota release:

Official Tasmota Release:
{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

vs.  Binary v1.3:
{"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5504,5536,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

jamesturton, 2020-12-16:
Hi @mgoeller!
That sounds exciting that you will be making your own ESP firmware. Hopefully I can help answer any questions you have.
I was actually working on a document defining the communication protocol used in this project which I have not quite finished writing yet, but the WIP version can be found here: [Communication.md](https://github.com/jamesturton/shelly-dimmer-stm32/blob/communication-protocol/Communication.md)

arendst actuall changed the GPIO design in Tasmota as I was developing the driver (see https://github.com/arendst/Tasmota/commit/9cbd2d531d646a14da110062c34ee21022f49ede) hence the GPIO number changing, but the pin function remains the same.

Looking at the pinout table on [Blakadder's template site](https://templates.blakadder.com/shelly_Dimmer_2.html) it is maybe easier to see what these pins do:
| GPIO # | Component        |
| ------ | ---------------- |
| GPIO00 | None             |
| GPIO01 | Serial Tx        |
| GPIO02 | None             |
| GPIO03 | Serial Rx        |
| GPIO04 | SHD Boot 0       |
| GPIO05 | SHD Reset        |
| GPIO09 | None             |
| GPIO10 | None             |
| GPIO12 | Switch2n         |
| GPIO13 | None             |
| GPIO14 | Switch1n         |
| GPIO15 | None             |
| GPIO16 | Led1i            |
| ADC0   | ADC Temperature  |

The pins `boot0` and `reset` are used to update the firmware in the STM32. To get the STM32 into serial bootloader mode you must assert the `reset` signal then pull `boot0` high and then release the `reset` signal. To reset to normal boot mode you can assert the `reset` signal then pull `boot0` low and then release the `reset` signal. 
You can check my implementation in the following functions in Tasmota `xdrv_45_shelly_dimmer.ino`: [`ShdResetToDFUMode()`](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_45_shelly_dimmer.ino#L111) and [`ShdResetToAppMode()`](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_45_shelly_dimmer.ino#L583).

mgoeller, 2021-01-08:
Thank you for your help! One more question and then I will close this issue, in your overview of the GPIOs I don't see that a HLW8012 power measurement IC is used for the Shelly Dimmer 2. I also don't see one on the PCB, how are you measuring the power then? And couldn't we use this to do a calibration procedure?

I already made a guess once in the original tasmota for shelly dimmer [thread](https://github.com/arendst/Tasmota/issues/6914#issuecomment-640190282):

"I would guess, that the calibration feature is using the power measurement combined with a dimming from 0-100%.

Knowing how much energy is used while slowly dimming up you can approximate the dimming curve of the lamp and map that to the linear "new" 0-100%.

E.g. a small LED lamp using the uncalibrated 0-100% only does a visual (/power consumption) change (and non linear!) from 1-15%. After calibration that is mapped more or less to the new 0-100% and allows much more granular control. And looks linear for the human eyes in brightness increase while sliding up."

And last question: do you know the GPIO for the physical reset button on the shelly dimmer 1/2?

jamesturton, 2021-01-12:
> Thank you for your help! One more question and then I will close this issue, in your overview of the GPIOs I don't see that a HLW8012 power measurement IC is used for the Shelly Dimmer 2. I also [...]

Power measurement on the dimmer 2 is done manually using two op-amps - one to measure voltage across the Live-Neutral and one to measure the voltage across a shunt resistor. This makes measuring the power a little bit more difficult as the ADCs must be polled multiple times during the cycle and the average value calculated.

I have been thinking a little about this and I think it would be best for the calibration values should be stored in the EEPROM used by the ESP8266 rather than the flash memory in the STM32 due to limited write endurance in the STM32.

> And last question: do you know the GPIO for the physical reset button on the shelly dimmer 1/2?

If I am not mistaken that is connected to the RST pin (physical pin 1, not GPIO 1) which is not assigned a GPIO number.

mgoeller, 2021-01-12:
Thank you James! The physical reset button on the Dimmer 2 is connected to GPIO 13, otherwise the STM would be reset every time it's pressed. Wiring it to GPIO13 has the advantage, that one then can define a long press or similar on the ESP and write your own reset routine (maybe even without reseting the STM, just restarting the ESP and so on).

And yes, I agree that the calibration values should be stored in the EEPROM for the ESP. I think that is also the case with Altercos original firmware.

closed 2021-01-12 by mgoeller