jamesturton/shelly-dimmer-stm32 #10: !Urgent! Dimmer 2 "Exploded" when uploading Firmware

Roxxor91, 2020-12-18:
Hi,
so this is serious. I do not know if this was your firmware (shelly-dimmer-stm32_v51.5.bin) or a coincidence.
I ordered a new shelly dimmer 2 and flashed tasmota via https://github.com/arendst/mgos-to-tasmota 
After that I uploaded like described here (https://templates.blakadder.com/shelly_Dimmer_2.html) the firmware file. When pushing, or maybe milliseconds after pressing the upload button I heard a loud pop and bad smell. I turned the main fuse off immediately and checked on the shelly. I attach pictures what I saw and smelled.......
I do not know if it is possible that the firmware led to this, or if it was pure coincidence. For now I guess I'm going to stick with the original firmware for the second dimmer I ordered. Not even sure if I want to trust the second device....
[image]

jamesturton, 2020-12-18:
Yes, this does look serious.
However it does seem like it could be a hardware problem, rather than a software problem... There have been many case of peoples units failing due to incorrect wiring posted on the [Facebook Shelly support group](https://www.facebook.com/groups/ShellyIoTCommunitySupport/). How was the unit wired? Was a load connected? Did you test the unit with stock firmware before trying this firmware?

Roxxor91, 2020-12-18:
Just to be sure it is not my own mistake. For setting the firmware up, i connected it to to N via the neural wire and to the "middle" L. There were only those two cables attached, no load, no switch. Was this a mistake?

jamesturton, 2020-12-18:
This looks very familiar to units that have been wired incorrectly.
Can you see which component might have failed?
Were you using single core wire or stranded wire? If using stranded wire, were you using ferrules? It is possible there may have a strand of the copper wire touching the 'O' output terminal from the 'N' neutral terminal causing a short circuit.

jamesturton, 2020-12-18:
FYI: I have now added a disclaimer to the readme document explaining that using this firmware is done under your own risk.

Roxxor91, 2020-12-18:
Stranded without ferrule, but I kept the exposed wire very short (about 3mm), so that even the isolation enters the Shelly, I specifically checked for those "stray" wires, as I knew they could lead to shorts.
Not sure what broke... I guess this "chip" in the middle?
[image]

jamesturton, 2020-12-19:
The 'chip' in the middle that seems burned is a small value, high power, resistor used for sensing the current. It's also possible to see that one of the pins (second from the left) connecting to the daughter board has actually burnt and been completely destroyed too. This indicates to me that there was a short circuit on the board. Is it possible for you to take a photo of the under side of the board?

Roxxor91, 2020-12-19:
I cant see any issues there.
[image]

jamesturton, 2020-12-20:
Yes, I can't see anything suspicious on the back side.
I'm not too sure what else can be done here... but here are some of my thoughts:

My best guess is that it there was a short circuit somewhere or one of the components failed (this device is rather cheap so it is possible that the cheaper components have poorer manufacturing tolerances). If there was something fundamentally wrong with this firmware then I would have expected to see this type of failure before. It works on mine and lots of other peoples devices without any problems.

I'm quite confident that even if the firmware upgrade to the STM32 somehow became corrupt, there is nothing in the hardware that would allow the device to 'self destruct' in this fashion. Of course it could be possible that I have missed something when reverse engineering this device.

Is it possible that Shelly have changed the design - maybe, the design and layout look the same, the markings at the top of the board look the same, but the date stamp is different
[image]
My dates from august. Hopefully this is just the date of manufacture and not a different version.

I am going to close this issue now, because like I said earlier, I don't think there is anything else I can do to help, but if you can think of anything you would like me to do then please let me know!

closed 2020-12-20 by jamesturton

Roxxor91, 2020-12-22:
Update: Also Shelly support confirmed it was an PCB defect. So I guess it was a coincidence that the update triggered it, or just after flashing something was activated which was shorted by the mentioned defect.