jamesturton/shelly-dimmer-stm32 #11: Dimmer Calibration / Dimming Type

Bascht74, 2020-12-20:
Hello James, 

I am right now switching to Tasmota and I have moved over amost all shelly 1/1PM/2.5/Switch S devices (26 done/22 to go). 8 of them are Dimmer (2x) / Dimmer 2 (6x). 

I would love to try, but I was looking after the "Dimming Type" (Trailing Edge / Leading Edge) Settings in Tasmota and I didn't find it so far. Is that implemented the you can set the Load type? (e.g. Leading Edge for LED-lights. 

At the same Spot is the calibration routine from Shelly. I don't know what it does, but usually the device doesn't make much noise afterwards if the lights are dimmed, and the Dimming Values a much more linear than before (I thing that is measures the load and then does a compensation that you will get linear movement in the load if you dim up).
Is that calibration inside Tasmota as well or is there a manual method to set a "cure" for the lights/lamps?

Last Question: Shelly added a "Anti-Flickering Debounce" with values 50 - 150 to one of the last firmware. Is that option accessible via Tasmota as well? 
Have you any experience with the "Ripple control signal" with your firmware ("Rundsteuerungs-Signal" in Germany, see https://en.wikipedia.org/wiki/Load_management)?

I haven't looked so far for min. brightness and warm-up possibilities, because want to check with you first. 

Maybe the information is interesting for other people and I could add it here: 
https://templates.blakadder.com/shelly_Dimmer_2.html

Regards, 

Sebastian

jamesturton, 2020-12-20:
Hi @Bascht74!
Thanks for taking an interest in the project!

> I would love to try, but I was looking after the "Dimming Type" (Trailing Edge / Leading Edge) Settings in Tasmota and I didn't find it so far. Is that implemented the you can set the Load type? (e.g. Leading Edge for LED-lights.

There is a special command in Tasmota: `ShdLeadingEdge`. `ShdLeadingEdge 1` will set to leading edge dimming and `ShdLeadingEdge 0` will set it to trailing edge dimming. The default is trailing edge dimming.

> At the same Spot is the calibration routine from Shelly. I don't know what it does, but usually the device doesn't make much noise afterwards if the lights are dimmed, and the Dimming Values a much [...]
Is that calibration inside Tasmota as well or is there a manual method to set a "cure" for the lights/lamps?

This has been mentioned before in the Tasmota issue https://github.com/arendst/Tasmota/issues/6914#issuecomment-647128683. Maybe you can see from the graph that it does indeed seem to make the power output more linear. My opinion is that a calibration routine like this should be implemented in Tasmota (or other firmware you choose to use on the ESP8266) as this would potentially be useful for other dimmers that Tasmota supports.

> Last Question: Shelly added a "Anti-Flickering Debounce" with values 50 - 150 to one of the last firmware. Is that option accessible via Tasmota as well?
Have you any experience with the "Ripple control signal" with your firmware ("Rundsteuerungs-Signal" in Germany, see https://en.wikipedia.org/wiki/Load_management)?

I have no idea what "Anti-Flickering Debounce" really means. We don't have a "Ripple control signal" here is it is impossible for me to test, but if you do in your area then it would be great if it is possible for you to test and report back with results!

> I haven't looked so far for min. brightness and warm-up possibilities, because want to check with you first.

Minimum brightness can be set with the [`DimmerRange`](https://tasmota.github.io/docs/Commands/#dimmerrange) command in Tasmota.

> Maybe the information is interesting for other people and I could add it here: https://templates.blakadder.com/shelly_Dimmer_2.html

That would be really great if you could update the templates website with this info! I'm sure other people would appreciate it too!

Bascht74, 2020-12-23:
Hi @jamesturton , 

thanks for the answers, I will try to add it to the documentation (submit an pull request).

I am still a little concerned about the migration of the dimmers to Tasmota, especially because of the ripple control signals in Germany. I had to set the debounce setting to 150 to get it right in the original Shelly firmware. 

So I sneaked around a bit because supporting that feature is possibly important for a log of people.
And I found something: 
https://github.com/Mollayo/Shelly-Dimmer-2/blob/master/light.cpp
@Mollayo has some code (`sendCmdSetDimmingParameters`) for setting these parameters. 
Maybe that can help getting the needed information?

Or might it be a good idea to get the serial output of the communication between the shelly firmware and the STM32?

Is it save to change to tasmota and load your firmware? If I need go back to the shelly firmware, does shelly install their own firmware again? (so that you can reverse the migration)

This guy has the same thing and I read something about calibration table, but I don't know weather it is handelt internally or in the STM32 (I am not a programmer).

And this guy has a calibration routine for shelly/STM32: 
https://gist.github.com/wichers/3c401f6dc6c61b4175ace7bc08cb6f8a

Sebastian

jamesturton, 2020-12-28:
I have talked with both Mollayo and Witchers when developing the Tasmota driver for the Shelly dimmers and they have taken slightly different approaches to what I have done.

Mollayo is making their own firmware for the ESP8266, rather than using Tasmota. They are also using the stock Shelly STM32 firmware, rather than using this open-source firmware for the STM32.

Witches has attempted to disassemble (reverse engineer) the stock firmware to decode the communication protocol for the calibration but I'm not sure if they managed to decode the calibration routine also? Maybe if @witchers is around they might be able to let us know where they got up to?

> Is it save to change to tasmota and load your firmware? If I need go back to the shelly firmware, does shelly install their own firmware again? (so that you can reverse the migration)

You can revert back to the stock firmware using the serial bootloader (if you have a USB to serial adapter). It might be possible to do this OTA using this tool https://github.com/yaourdt/tasmota-to-mgos but I have never tried. Good luck!

It sounds like I will have to make some changes to how this firmware works to support the dimmer 2 with no neutral so I will add some filtering to cope with the ripple control signals while I'm at it.

CReimer, 2021-01-06:
I can confirm that tasmota-to-mgos works. I had to move all my dimmers back to stock firmware. Will try Tasmota again when #7 is fixed

bakeromso, 2021-03-29:
First off, many thanks for all your work on the project @jamesturton! I was wondering if you would know if there is any progress on the calibration routine from withing Tasmota? 
Also I believe you meant to tag @wichers in your post before, although I'm not sure if they will get a notification if they been involved in this Github issue.

jamesturton, 2021-05-02:
Closing this issue as further discussion regarding calibration should be made on Tasmota repository.
An attempt to prevent flickering with 'Rundsteuerungs-Signal' has been commited in 607c22d. @Bascht74 please make a new issue if the 'Rundsteuerungs-Signal' is still causing issues.

closed 2021-05-02 by jamesturton

Bascht74, 2021-05-02:
@Bascht74: Thanks! I switched my first Shelly Dimmer 1 to Tasmota with the new firmware and right now, it seems fine. Let's see tomorrow if any issues come along. If everything is fine, I won't make a new issue and if there are still problems, I make a new issue as you said. Great work!
Sebastian

BTW: Do you have any reference for the calibration topic at Tasmota? Is there something planned...?

victorhooi, 2022-04-27:
Does anybody know if Tasmota ended up adding calibration support for the Shelly Dimmer 2?

Or is there some way to do it with existing functionality?

bakeromso, 2022-04-27:
I havent found a way to do this yet, curious to learn this as well!

jamesturton, 2022-04-27:
> Does anybody know if Tasmota ended up adding calibration support for the Shelly Dimmer 2?
> 
> Or is there some way to do it with existing functionality?

Are you talking about gamma correction? This is already implemented in Tasmota: https://tasmota.github.io/docs/Lights/#gamma-correction
It's also included in the instructions for setting-up the Shelly dimmer 2 on blakadder's wiki page: https://templates.blakadder.com/shelly_Dimmer_2.html

bakeromso, 2022-04-27:
I think this is about correcting for calibrating so that the luminous intensity (or say our perception thereof) increases linearly with the linear increment on the dimmer value. So on turning the knob 10%, the luminous intensity would change with 10% as well.

For me, at the moment there is a very sensitive response when going from almost fully dimmed to a bit less dimmed, compared to the mid range when there is little noticable difference for changing dimming values

bakeromso, 2022-04-27:
Thanks for the suggestion! Both were no help for me. Changing the range does not change the behavior of a mismatched relation (non linear relation) between value increase and intensity

victorhooi, 2022-04-29:
Wait - can anybody clarify - is gamma correction and dimmer range the same as what Shelly does in the official firmware, when you click "Calibrate"?

Or is there something else that needs to happen to "calibrate" a dimmer?

ESPHome does offer a gamma_correct argument ([reference](https://esphome.io/components/light/index.html
)), but I can't see anything for dimmer range.

victorhooi, 2022-05-07:
@jamesturton Thanks for the link, with the explanation.

You mentioned you were able to measure the curve.

However, do you have any ideas for how to calibrate it (i.e. get our curve more linear)? Either for Tasmota or ESPHome?

Also - with the DimmerRange - how do you actually work out what the correct range for a device/lamp should be?

Finally - somewhat unrelated - just curious - do you have more stuff on the roadmap or backlog for this firmware? =)

glyndon, 2022-11-06:
Seems like calibration has more to do with the current draw (or some other electrical characteristic of the load) at various dimming levels.
It couldn't be related to the brightness curbe, since the Shelly can't _see_ how bright the lamp is at any given electrical dimming level.

Ara-Nova, 2022-12-14:
unfortunately my dimmer only works, when im using ShdLeadingEdge 1 first and then ShdLeadingEdge 0.

after a few switch tapping my Shelly Dimmer 2 isnt working anmymore. i have to reenter ShdLeadingEdge 1 and 0 . The config will not be saved. savedata0 and savedata1 not working.

Please help