jamesturton/shelly-dimmer-stm32 #12: Shelly Dimmer 1 - Tasmota configuration

luke-barnett, 2020-12-23:
I have successfully flashed a shelly dimmer 1 module and gotten it set up and configured mostly to what I need but I have a couple of questions. These may be more tasmota related and in that case, I might be asking in the wrong place so please feel free to push me along.

- I have the dimmer behind a mechanical toggle switch. But I cannot get the toggle behaviour that I want. I want it to simply turn on and off the light as a normal switch would do. Instead on a toggle of the switch, it pulses the light on and off, presumably as a mechanism to set the brightness? Sometimes you can time it right and get it to turn on/off as you want it to but ideally, the switch just toggles the lights as desired.
- I've set a dimmer range, and so can dim the lights based on their range, but the shelly itself is very noisy when dimmed, to the point that the dimming function practically wouldn't be used. Is there anything I can do to reduce the noise? It does trail off with higher brightness.

rbswift, 2020-12-24:
1. Does switching the dimmer on/off work as expected using the Toggle button in the tasmota web interface? If it does then you possibly just need to set `SwitchMode 0` in Tasmota. If that doesn't work maybe post your tasmota log.
2. I've noticed a slight hum from the dimmer too. I am by no means expert here but check your tasmota log to be sure you're running the latest version "_SHD: Shelly Dimmer Co-processor Version v51.5_" and use `ShdLeadingEdge 0` (default) if possible.

luke-barnett, 2020-12-24:
@rbswift thanks, ShdLeadingEdge solved the noise issue 👍 

Here are some logs from the console with `WebLog 3` switching it once, which resulted in the lights flashing, and then once again to stop it from flashing. I have it wired like this diagram:

[image]

> 02:53:32 SRC: Switch
> 02:53:32 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:32 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:32 SRC: Switch
> 02:53:32 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:32 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:32 SRC: Switch
> 02:53:32 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:32 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:32 CFG: Saved to flash at F4, Count 1079, Bytes 4096
> 02:53:33 SRC: Switch
> 02:53:33 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:33 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:33 CFG: Saved to flash at FB, Count 1080, Bytes 4096
> 02:53:34 SRC: Switch
> 02:53:34 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:34 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:34 CFG: Saved to flash at FA, Count 1081, Bytes 4096
> 02:53:35 SRC: Switch
> 02:53:35 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:35 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:35 CFG: Saved to flash at F9, Count 1082, Bytes 4096
> 02:53:36 SRC: Switch
> 02:53:36 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:36 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:36 CFG: Saved to flash at F8, Count 1083, Bytes 4096
> 02:53:37 SRC: Switch
> 02:53:37 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:37 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:37 CFG: Saved to flash at F7, Count 1084, Bytes 4096
> 02:53:38 SRC: Switch
> 02:53:38 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:38 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:38 CFG: Saved to flash at F6, Count 1085, Bytes 4096
> 02:53:39 SRC: Switch
> 02:53:39 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:39 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:39 CFG: Saved to flash at F5, Count 1086, Bytes 4096
> 02:53:40 SRC: Switch
> 02:53:40 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:40 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:40 CFG: Saved to flash at F4, Count 1087, Bytes 4096
> 02:53:41 SRC: Switch
> 02:53:41 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:41 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:41 CFG: Saved to flash at FB, Count 1088, Bytes 4096
> 02:53:42 SRC: Switch
> 02:53:42 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"OFF"}
> 02:53:42 MQT: stat/tasmota_D17AE1/POWER = OFF
> 02:53:42 CFG: Saved to flash at FA, Count 1089, Bytes 4096
> 02:53:42 SRC: Switch
> 02:53:42 MQT: stat/tasmota_D17AE1/RESULT = {"POWER":"ON"}
> 02:53:42 MQT: stat/tasmota_D17AE1/POWER = ON
> 02:53:43 CFG: Saved to flash at F9, Count 1090, Bytes 4096

luke-barnett, 2020-12-24:
Had the wiring incorrect, corrected and everything is working as expected. A classic case of always needing to buzz it out and not make assumptions.

closed 2020-12-24 by luke-barnett