jamesturton/shelly-dimmer-stm32 #13: Flashing to Tasmota 9.2.0 with Shelly Dimmer 2

tfcarlin, 2020-12-27:
Hi,

I'm following these instructions for a Shelly Dimmer 2 but might be misunderstanding:
After flashing Tasmota, apply the template and then use the Firmware upgrade - File upload menu to flash the required STM32 firmware (provided at shelly-dimmer-stm32 releases page) to the MCU. The template must be set before flashing the STM32 firmware.

Using the Tasmota UI I get:
Upload Failed
Invalid file signature

I see this on the serial interface:
`01:11:14 QPC: Reset
01:11:28 UPL: File shelly-dimmer-stm32_v51.5.bin ...
01:12:16 CMD: template
01:12:16 RSL: stat/tasmota_0071D1/RESULT = {"NAME":"ShellyDimmer2","GPIO":[0,3200,1,3232,5568,5600,0,0,193,0,192,1,320,4736],"FLAG":0,"BASE":18}
01:12:32 UPL: File shelly-dimmer-stm32_v51.5.bin ...
`

Any suggestions?

Thanks

rbswift, 2020-12-27:
steps to try:

1. install tasmota 9.2 full (not minimal) build from http://ota.tasmota.com/tasmota/release/tasmota.bin.gz
2. update template from https://templates.blakadder.com/shelly_Dimmer_2.html
3. restart
4. upload firmware from https://github.com/jamesturton/shelly-dimmer-stm32/releases/download/v51.5/shelly-dimmer-stm32_v51.5.bin

then post your web console log here. good luck

tfcarlin, 2020-12-27:
@rbswift Thanks for the suggestions. Those are the steps I have been following.

Here is the system Info:

[image]

I use the "Upgrade by file upload" UI:

[image]

The template appears to be correct:

17:03:27.591 RSL: stat/tasmota_0071D1/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

And I continue to receive:

[image]

with the output in the serial interface:

17:04:44.376 CFG: Saved to flash at FB, Count 15, Bytes 4096
17:04:44.376 UPL: File shellystm.bin ...
17:04:44.456 HTP: Upload done

rbswift, 2020-12-27:
The template looks wrong. The console would show Shelly Dimmer instead of Tasmota after the template is set.

In the web console enter:

Template {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

overflywingo, 2020-12-28:
Hello, I do have exactly the same problem ... with Sonoff basic Module still displayed as web ui and Upload Failed due to Invalid file signature  although I entered the Template via the web console

Program Version | 9.2.0(tasmota)
-- | --
2020-12-21T15:03:40
2_7_4_9/2.2.2-dev(38a443e)
0T00:04:48
14 at 0xFB000
6
Software/System restart
Tasmota
 
EFLA_Hall (48%, -76 dBm)
tasmota_9666AF-1711
E0:98:06:96:66:AF
192.168.1.193
 
192.168.1.1
255.255.255.0
192.168.1.1
 
 
1883
DVES_USER
DVES_9666AF
tasmota_%06X
cmnd/tasmotas/
cmnd/tasmota_9666AF/
cmnd/DVES_9666AF_fb/
Disabled
 
None
Disabled
 
9856687
0x15701C
2048kB
1024kB
586kB
416kB
25kB

rbswift, 2020-12-28:
restart the device, type template in the web `template` and then copy/paste the entire web console output back here. I'm no expert but can try to help if nobody else can

overflywingo, 2020-12-28:
Hi thanks , here it is : 
09:43:35 CMD: template
09:43:35 RSL: stat/tasmota_9666AF/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

BTW the file that I'm trying to flash is named : shelly-dimmer-stm32_v51.5.bin (and not shellystm.bin) but I suppose it's OK

rbswift, 2020-12-28:
"entire web console output" please ... not just two lines

overflywingo, 2020-12-28:
Oups ! Sorry 

00:00:00 CFG: Loaded from flash at FB, Count 16
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: Connecting to AP1 EFLA_Hall Channel 13 BSSId 18:E8:29:9D:4D:D9 in mode 11n as tasmota_9666AF-1711...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:00:22 RSL: tele/tasmota_9666AF/INFO1 = {"Module":"Sonoff Basic","Version":"9.2.0(tasmota)","FallbackTopic":"cmnd/DVES_9666AF_fb/","GroupTopic":"cmnd/tasmotas/"}
10:00:22 RSL: tele/tasmota_9666AF/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_9666AF-1711","IPAddress":"192.168.1.193"}
10:00:22 RSL: tele/tasmota_9666AF/INFO3 = {"RestartReason":"Software/System restart"}
10:00:22 RSL: stat/tasmota_9666AF/RESULT = {"POWER":"OFF"}
10:00:22 RSL: stat/tasmota_9666AF/POWER = OFF
10:00:25 QPC: Reset
10:00:26 RSL: tele/tasmota_9666AF/STATE = {"Time":"2020-12-28T10:00:26","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EFLA_Hall","BSSId":"18:E8:29:9D:4D:D9","Channel":13,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:00:37 CMD: template
10:00:37 RSL: stat/tasmota_9666AF/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

rbswift, 2020-12-28:
that all looks fine. I have a Shelly Dimmer 1 but not much difference. Can you post the entire console log including an attempt to load the firmware? You should see something towards the end like this briefly before it restarts....

```
09:05:29 HTP: Firmware Upgrade
09:05:49 HTP: Main Menu
09:05:51 HTP: Console
09:05:57 CFG: Saved to flash at FB, Count 748, Bytes 4096
09:05:57 UPL: File shelly-dimmer-stm32_v51.5.bin ...
09:05:57 UPL: File type 5
09:05:57 UPL: Transfer 13944 bytes
09:05:59 UPL: Successful 13944 bytes
09:05:59 HTP: Upload done
09:05:59 SRC: WebGui from 192.168.188.21
```

overflywingo, 2020-12-28:
00:00:00 CFG: Loaded from flash at FB, Count 20
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: Connecting to AP1 EFLA_Hall Channel 13 BSSId 18:E8:29:9D:4D:D9 in mode 11n as tasmota_9666AF-1711...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:09:42 RSL: tele/tasmota_9666AF/INFO1 = {"Module":"Sonoff Basic","Version":"9.2.0(tasmota)","FallbackTopic":"cmnd/DVES_9666AF_fb/","GroupTopic":"cmnd/tasmotas/"}
10:09:42 RSL: tele/tasmota_9666AF/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_9666AF-1711","IPAddress":"192.168.1.193"}
10:09:42 RSL: tele/tasmota_9666AF/INFO3 = {"RestartReason":"Software/System restart"}
10:09:42 RSL: stat/tasmota_9666AF/RESULT = {"POWER":"OFF"}
10:09:42 RSL: stat/tasmota_9666AF/POWER = OFF
10:09:45 QPC: Reset
10:09:46 RSL: tele/tasmota_9666AF/STATE = {"Time":"2020-12-28T10:09:46","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EFLA_Hall","BSSId":"18:E8:29:9D:4D:D9","Channel":13,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:10:11 UPL: File shelly-dimmer-stm32_v51.5.bin ...

then nothing on the console but the message in a popup "Upload Failed due to Invalid file signature"

overflywingo, 2020-12-28:
Hi, I tried also with OTA URL
00:00:00 CFG: Loaded from flash at FB, Count 20
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: Connecting to AP1 EFLA_Hall Channel 13 BSSId 18:E8:29:9D:4D:D9 in mode 11n as tasmota_9666AF-1711...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:09:42 RSL: tele/tasmota_9666AF/INFO1 = {"Module":"Sonoff Basic","Version":"9.2.0(tasmota)","FallbackTopic":"cmnd/DVES_9666AF_fb/","GroupTopic":"cmnd/tasmotas/"}
10:09:42 RSL: tele/tasmota_9666AF/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_9666AF-1711","IPAddress":"192.168.1.193"}
10:09:42 RSL: tele/tasmota_9666AF/INFO3 = {"RestartReason":"Software/System restart"}
10:09:42 RSL: stat/tasmota_9666AF/RESULT = {"POWER":"OFF"}
10:09:42 RSL: stat/tasmota_9666AF/POWER = OFF
10:09:45 QPC: Reset
10:09:46 RSL: tele/tasmota_9666AF/STATE = {"Time":"2020-12-28T10:09:46","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EFLA_Hall","BSSId":"18:E8:29:9D:4D:D9","Channel":13,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:10:11 UPL: File shelly-dimmer-stm32_v51.5.bin ...
10:14:46 RSL: tele/tasmota_9666AF/STATE = {"Time":"2020-12-28T10:14:46","Uptime":"0T00:05:09","UptimeSec":309,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EFLA_Hall","BSSId":"18:E8:29:9D:4D:D9","Channel":13,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:15:03 RSL: stat/tasmota_9666AF/RESULT = {"OtaUrl":"https://github.com/jamesturton/shelly-dimmer-stm32/releases/download/v51.5/shelly-dimmer-stm32_v51.5.bin"}
10:15:03 RSL: stat/tasmota_9666AF/RESULT = {"Upgrade":"Version 9.2.0 from https://github.com/jamesturton/shelly-dimmer-stm32/releases/download/v51.5/shelly-dimmer-stm32_v51.5.bin"}
10:15:05 HTP: Web server stopped
10:15:06 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:15:07 HTP: Web server stopped
10:15:08 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:15:09 HTP: Web server stopped
10:15:10 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:15:11 HTP: Web server stopped
10:15:12 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:15:13 HTP: Web server stopped
10:15:14 HTP: Web server active on tasmota_9666AF-1711 with IP address 192.168.1.193
10:15:15 RSL: stat/tasmota_9666AF/UPGRADE = {"Upgrade":"Failed HTTP error: connection lost"}

rbswift, 2020-12-28:
I'm not sure what else to suggest sorry. Maybe check the file you've downloaded is exactly 13944 bytes. I've also posted the first few bytes in hex so you can check no corruption but that's unlikely. If your device is still showing Sonoff then it's like the template isn't taking properly. Maybe reset all the configuration and install tasmota again from scratch?

```
% ls -al shelly-dimmer-stm32_v51.5.bin 
-rw-r--r--@ 1 xx  xx  13944 28 Dec 20:04 shelly-dimmer-stm32_v51.5.bin
% hexdump shelly-dimmer-stm32_v51.5.bin 
0000000 00 10 00 20 d1 15 00 08 cd 15 00 08 cb 15 00 08
0000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0000020 00 00 00 00 00 00 00 00 00 00 00 00 cd 15 00 08
0000030 00 00 00 00 00 00 00 00 cd 15 00 08 7d 08 00 08
0000040 cb 15 00 08 cb 15 00 08 cb 15 00 08 cb 15 00 08
0000050 cb 15 00 08 cb 15 00 08 b5 0a 00 08 cb 15 00 08
0000060 cb 15 00 08 cb 15 00 08 cb 15 00 08 cb 15 00 08
0000070 cb 15 00 08 cb 15 00 08 8d 08 00 08 15 09 00 08
0000080 25 0a 00 08 cb 15 00 08 cb 15 00 08 cb 15 00 08
0000090 cb 15 00 08 cb 15 00 08 cb 15 00 08 cb 15 00 08
00000a0 cb 15 00 08 cb 15 00 08 cb 15 00 08 2d 01 00 08
00000b0 cb 15 00 08 cb 15 00 08 cb 15 00 08 cb 15 00 08

```

rbswift, 2020-12-28:
Just noticed you've got module 1. You might need to try to set `module 0`  in the console?

overflywingo, 2020-12-28:
YOU SAVED MY DAY :)   THANK YOU VERY VERY MUCH

rbswift, 2020-12-28:
You're welcome. Tricky one - I missed it a couple of times! Good to know I've still got some troubleshooting skills left in me! Enjoy James' hard work!

jamesturton, 2020-12-28:
Huge thanks @rbswift for all of your hard work solving these problems!

closed 2020-12-28 by jamesturton