jamesturton/shelly-dimmer-stm32 #15: Can't get Dimmer2 to dim at all with wall switches, no matter what switchmode type

rkhanso, 2021-01-03:
I have the new Shelly Dimmer2 devices. I flashed to Tasmota.bin first, then shelly-dimmer-stm32_v51.5.bin
I used your example template {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}

The Shelly will Dim using the Shelly/Tasmota web GUI and also within Home Assistant.  But I can't get it to dim using the two momentary wall switches connected to the Shelly (one to SW1, the other to SW2 terminals).  

I've tried every switchmode in the Console and can get both switches to turn the lights on/off using
switchmode1 6
switchmode2 6

But when I try switchmode 11, the light switches don't do a thing. Don't dim, don't turn off/on.

I'm a beginner with flashing/Tasmota/Home Assistant, etc - so it certainly could be something I've done wrong. But I know the switchmode options aren't doing what I expect they should do.

Thanks.

BPSoft, 2021-01-03:
Try with this SetOptions:
Backlog LedTable 0; dimmerstep 5; DimmerRange 15,100; SwitchMode1 11; SwitchMode2 11; SwitchDebounce 40; SetOption32 1; SetOption13 1; SetOption99 1; SetOption93 1; SetOption40 1; SetOption34 1; SetOption1 1

Disable all active rules, and use this instead:
Rule1
  ON switch1#state DO cmnd/%topic%/DIMMER + BREAK
  ON Dimmer#State=1 DO Backlog cmnd/%topic%/DIMMER 2; cmnd/%topic%/POWER OFF BREAK 
  ON switch2#state DO cmnd/%topic%/DIMMER - ENDON
Rule1 1

Don't forget to change %topic% by yours, or use alternative Rule1:
Rule1
  ON switch1#state DO DIMMER + BREAK
  ON Dimmer#State=1 DO Backlog DIMMER 2; POWER OFF BREAK 
  ON switch2#state DO DIMMER - ENDON   
Rule1 1

jamesturton, 2021-01-04:
Thanks @BPSoft for helping @rkhanso with this! Hopefully this has solved their problem. I'm going to close this issue now as this problem with the configuration with Tasmota - not this firmware. If you need any more help, I would suggest making a new issue on the Tasmota repo.

closed 2021-01-04 by jamesturton

rkhanso, 2021-01-04:
> Try with this SetOptions:
> Backlog LedTable 0; dimmerstep 5; DimmerRange 15,100; SwitchMode1 11; SwitchMode2 11; SwitchDebounce 40; SetOption32 1; SetOption13 1; SetOption99 1; SetOption93 1; [...]

That didn't fix it. 
Still no dimming with wall switches. Only the web/GUI and Home Assistant.

BPSoft, 2021-01-04:
OK, please describe your Wall Switches and physical connection with the shelly. In particular define L and N in relation to Wall Switch and Shelly SW1 and SW2. It is/was working with original/stock Shelly firmware?

rkhanso, 2021-01-04:
Hello.

Thanks for the reply/follow-up.

Everything was working with the Shelly firmware that was on the Dimmer2 
out of the package.  Short press on either switch turned the lights 
on/off. Long press either switch dimmed the lights. I'm also pretty sure 
that double-tapping brought lights to full brightness.

With the Tasmota firmware, I can only get the on/off function working 
(switchmode 6) on both of the wall switches. Wall switches are rocker 
switches converted to momentary switches.

About the wiring:
L and N are in the ceiling light fixture, so that 's where I put the 
Shelly. And the Shelly is powered from them, pig-tailed off each bundle.

Each of the two wall switches have a green (or maybe it's blue) wire 
going to the ceiling fixture. I connected a green/blue wire from one 
switch to the SW1 terminal on the Shelly. The other green/blue wire to 
the other wall switch is on SW2 on the Shelly. There is a yellow wire 
going directly from one switch to the other.  Each wall switch also has 
a Hot/L wire to it.  There is no N/Neutral wire connected to either wall 
switch.   I know these are working correctly because when I use 
SwitchMode1 6 and SwitchMode2 6, and press the switches, the lights turn 
on/off like they should.

Output from the Shelly goes to the black wire on the light fixture. All 
the N/Neutral wires are tied together in the fixture box, including to 
the light.

The fundamentals are....when using Switchmode 6, the lights operate like 
they should - a simple toggle on/off. It doesn't matter what wall switch 
you use. Press one switch, lights go on. Press the same wall switch or 
the other switch, lights go off. And vice versa.

When I change to SwitchMode12, the wall switches do nothing at all, 
whether sort-pressed, long-pressed or double-tapped. The lights just 
stay in the state they started (either on or off).

I'm such a novice that I don't know if this means it's a Tasmota problem 
(either firmware, Template, SetOptions, rules, etc), or a STM Firmware 
problem (but I suspect not since dimming works OK using the web/GUI), or 
something else. Or if more troubleshooting needs to be done to determine 
what the problem is.

That's where I'm at now. I'd like to get the dimming working again, but 
with Tasmota. Also a double-tap to reset to 100% or to change the 
dimming direction - whatever fancy feature.   If not possible, I'm 
wondering if I'd be able to put the Shelly firmware back on and get it 
to stock out of the package and working like before.

Thanks.

Roger

BPSoft, 2021-01-04:
I have an aditional Shelly Dimmer 2 that i can test with a similar setup and i will report you back as soon as possible.

> When I change to SwitchMode12, the wall switches do nothing at all, whether sort-pressed, long-pressed or double-tapped. The lights just stay in the state they started (either on or off).

Please check the output in the Tasmota Web Console while trying the Switches, because the SetOptions i gave you are doing DIMMER + in SW1 and DIMMER - in SW2. So If you are in Zero what could happen is that you have to do multiple momentary Switches UP until Light Goes ON, it could happen at 16, 20, 30, it is Lamp dependable. What kind of Lamp are you using?

Try also with the smallest momentary switch you can and check de Console for some kind of response

rkhanso, 2021-01-04:
To be sure I'm doing what you request correctly....

Should I do *SO114 1*   to get the switch data to show up in the 
console?  Or, how do I accomplish that?

I'm using LED track lighting, with THESE 
<https://www.menards.com/main/electrical/light-bulbs/led-light-bulbs/feit-electric-50w-equivalent-mr16-gu10-led-light-bulb-6-pack/mr16gu10500-10kled-6/p-1543822036656-c-7482.htm?tid=6486577907567190694&ipos=8>bulbs.  
Feit LED dimmable LED replacements for Halogen. The track lighting has 6 
of those lights on it.

I'm not sure what you mean by the statement "Try also with the smallest 
momentary switch you can and check de Console for some kind of response"

BPSoft, 2021-01-05:
> Should I do *SO114 1*   to get the switch data to show up in the console?  Or, how do I accomplish that?

No, don't turn on SO114, keep it SO114 0
You shouldn't do anything more than open Console, where you already should receive something like this from time to time (5mins):

23:28:39 RSL: tele/tasmota_XXXXX/STATE = {"Time":"2021-01-04T23:28:39","Uptime":"0T17:14:05","UptimeSec":62045...
23:28:39 RSL: tele/tasmota_XXXXX/SENSOR = {"Time":"2021-01-04T23:28:39","ANALOG":{"Temperature":66.6},"ENERGY"...

When the rule is triggered you should receive something like:
00:49:09 RSL: stat/tasmota_XXXXXX/RESULT = {"POWER":"ON","Dimmer":40}

That indicates the new values for Power and Dimmer.

> I'm not sure what you mean by the statement "Try also with the smallest momentary switch you can and check de Console for some kind of response"

Just try to press the switch for the minimum time you can.

Give me 10 minutes, and i will try to give a better set of rules to test

BPSoft, 2021-01-05:
Ok please try this:

Backlog SwitchMode1 11; SwitchMode2 11; SetOption32 8

Rule1
ON switch1#state=2 DO cmnd/%topic%/POWER ON ENDON
ON switch1#state=4 DO cmnd/%topic%/DIMMER + ENDON
Rule1 1

Rule2
ON switch2#state=2 DO cmnd/%topic%/POWER OFF ENDON
ON switch2#state=4 DO cmnd/%topic%/DIMMER - ENDON
Rule2 1

Don't forget to change %topic% with yours, it should be something like this "tasmota_XXXXX"

rkhanso, 2021-01-05:
OK - I put your options and rules in. I found this on the info page:

MQTT Full Topic 	cmnd/tasmota_0042DD/

So this is what I entered in the console  --  but then I remembered that 
I'm pretty sure I need switchmode 12 for inverted (since I need 
switchmode 6 for basic on/off to work correctly):

01:30:57 CMD: Backlog SwitchMode1 11; SwitchMode2 11; SetOption32 8
01:30:57 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode1":11}
01:30:57 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode2":11}
01:30:57 MQT: stat/tasmota_0042DD/RESULT = {"SetOption32":8}
01:34:00 CMD: Backlog SwitchMode1 11; SwitchMode2 11; SetOption32 8
01:34:00 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode1":11}
01:34:00 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode2":11}
01:34:00 MQT: stat/tasmota_0042DD/RESULT = {"SetOption32":8}
01:34:05 MQT: tele/tasmota_0042DD/STATE = 
{"Time":"2021-01-05T01:34:05","Uptime":"0T06:20:11","UptimeSec":22811,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":61,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"IOT","BSSId":"DC:9F:DB:B4:2F:01","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
01:34:05 MQT: tele/tasmota_0042DD/SENSOR = 
{"Time":"2021-01-05T01:34:05","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":116.2},"ENERGY":{"TotalStartTime":"2021-01-04T15:30:52","Total":0.007,"Yesterday":0.007,"Today":0.000,"Period":0,"Power":30},"TempUnit":"F"}
01:34:08 CMD: Rule1 ON switch1#state=2 DO cmnd/tasmota_0042DD/POWER ON 
ENDON ON switch1#state=4 DO cmnd/tasmota_0042DD/DIMMER + ENDON
01:34:08 RUL: Stored uncompressed, would compress from 113 to 67 (-41%)
01:34:08 MQT: stat/tasmota_0042DD/RESULT = 
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":113,"Free":398,"Rules":"ON 
switch1#state=2 DO cmnd/tasmota_0042DD/POWER ON ENDON ON switch1#state=4 
DO cmnd/tasmota_0042DD/DIMMER + ENDON"}
01:34:10 CMD: rule1 1
01:34:10 MQT: stat/tasmota_0042DD/RESULT = 
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":113,"Free":398,"Rules":"ON 
switch1#state=2 DO cmnd/tasmota_0042DD/POWER ON ENDON ON switch1#state=4 
DO cmnd/tasmota_0042DD/DIMMER + ENDON"}
01:34:17 CMD: Rule2 ON switch2#state=2 DO cmnd/tasmota_0042DD/POWER OFF 
ENDON ON switch2#state=4 DO cmnd/tasmota_0042DD/DIMMER - ENDON
01:34:17 RUL: Stored uncompressed, would compress from 114 to 69 (-39%)
01:34:17 MQT: stat/tasmota_0042DD/RESULT = 
{"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":114,"Free":397,"Rules":"ON 
switch2#state=2 DO cmnd/tasmota_0042DD/POWER OFF ENDON ON 
switch2#state=4 DO cmnd/tasmota_0042DD/DIMMER - ENDON"}
01:34:19 CMD: rule2 1
01:34:19 MQT: stat/tasmota_0042DD/RESULT = 
{"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":114,"Free":397,"Rules":"ON 
switch2#state=2 DO cmnd/tasmota_0042DD/POWER OFF ENDON ON 
switch2#state=4 DO cmnd/tasmota_0042DD/DIMMER - ENDON"}

01:35:51 CMD: switchmode1 12
01:35:51 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode1":12}
01:35:55 CMD: switchmode2 12
01:35:55 MQT: stat/tasmota_0042DD/RESULT = {"SwitchMode2":12}

PROGRESS --
I pressed the switches.  With a long-press, SW1 dims UP.  With a 
long-press, SW2 dims DOWN.
quick, single presses don't do anything.

The dimming Up or Down is in steps, not smooth.

If you want some output, here is some. Let me know if you want me to do 
something in particular.

01:38:45 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:38:45 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":100}
01:38:45 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:38:45 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":100}
01:38:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:38:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":100}
01:38:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:38:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":100}
01:38:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:38:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":100}
[... 201 lines cut ...]
01:39:40 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":1}
01:39:40 RUL: SWITCH2#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER -"
01:39:40 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":1}
01:39:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":6}
01:39:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":11}
01:39:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":16}
01:39:47 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:47 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":21}
01:39:47 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:47 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":26}
01:39:47 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:47 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":31}
01:39:48 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:48 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":36}
01:39:48 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:48 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":41}
01:39:48 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:48 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":46}
01:39:49 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:49 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":51}
01:39:49 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
01:39:49 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":56}

BPSoft, 2021-01-05:
Nice!

It's not smooth because DimmerStep 5. Do:

DimmerStep 1

and try again

BPSoft, 2021-01-05:
Also, if you want to swap the Switches, in the rules change switch1#state with switch2#state and vice versa.

rkhanso, 2021-01-05:
That helped some. It's still a little jerky going brighter.

It seems like the SW2 (off/dim switch) is almost working like I want it.
If the lights are on and I long-press, the lights start to dim. If the 
lights are on and I quick-tap SW2, the lights go off. I can long-press 
SW2 and the lights will go up to the brightness they were at and then 
start to dim again.

SW1 doesn't do anything with a quick-tap.  If I hold it, it'll brighten 
the lights.

rkhanso, 2021-01-05:
I'd really like both switches to work the same.  And like they did with 
the Shelly firmware.

    Quick press on either switch will toggle the lights. Like a
    traditional 3-way setup.
    Long-press on either switch will dim the lights.
    Double-tap on either switch will go to full-bright.   Unless it's
    possible to through some other way, alternate a long-press to go up
    or down in brightness.

Is this possible?

I could mess around with it if I knew what to do - hoping I could get it 
to do what my end-goal is.

BPSoft, 2021-01-05:
Yes it's possible, with something like this:

Rule1
ON System#boot mem1 + ENDON
ON switch1#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON
ON switch1#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON

Rule1 1

Rule2
ON System#boot mem1 + ENDON
ON switch2#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON
ON switch2#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON
ON switch2#state=5 DO mem1 - ENDON
ON switch2#state=6 DO mem1 + ENDON

Rule2 1

rkhanso, 2021-01-05:
Thanks for the help getting this far. I really appreciate it.

When I use those rules (I did put in tasmota_0042DD  in place of the 
%topic%), there is only "dimming" in the BRIGHTER direction on both 
switches. Neither switch will lower the brightness. Only raise it if 
it's not at 100%.

FYI - I first pasted them in without changing the %topic% and it did the 
same thing.

Double-tapping doesn't do anything.

Here is some output of single quick taps, long-presses and double tap at 
the end.

03:37:20 CMD: dimmerstep 2
03:37:20 MQT: stat/tasmota_0042DD/RESULT = {"DimmerStep":2}
03:37:36 RUL: SWITCH2#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:36 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":33}
03:37:36 RUL: SWITCH2#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:36 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":35}
03:37:36 RUL: SWITCH2#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:36 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":37}
03:37:37 RUL: SWITCH2#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:37 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":39}
03:37:37 RUL: SWITCH2#STATE=6 performs "mem1 +"
03:37:37 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:38 RUL: SWITCH2#STATE=5 performs "mem1 -"
03:37:38 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"-"}
03:37:38 RUL: SWITCH2#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:38 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"OFF"}
03:37:38 MQT: stat/tasmota_0042DD/POWER = OFF
03:37:39 RUL: SWITCH2#STATE=6 performs "mem1 +"
03:37:39 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:40 RUL: SWITCH2#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:40 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON"}
03:37:40 MQT: stat/tasmota_0042DD/POWER = ON
03:37:41 RUL: SWITCH2#STATE=6 performs "mem1 +"
03:37:41 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:45 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:45 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":41}
03:37:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":43}
03:37:46 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:46 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":45}
03:37:47 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:47 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:48 RUL: SWITCH1#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:48 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"OFF"}
03:37:48 MQT: stat/tasmota_0042DD/POWER = OFF
03:37:49 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:49 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:49 RUL: SWITCH1#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:49 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON"}
03:37:49 MQT: stat/tasmota_0042DD/POWER = ON
03:37:50 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:50 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:52 RUL: SWITCH1#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:52 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"OFF"}
03:37:52 MQT: stat/tasmota_0042DD/POWER = OFF
03:37:52 RUL: SWITCH1#STATE=5 performs "mem1 -"
03:37:52 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"-"}
03:37:53 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:53 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:54 RUL: SWITCH1#STATE=2 performs "cmnd/tasmota_0042DD/POWER TOGGLE"
03:37:54 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON"}
03:37:54 MQT: stat/tasmota_0042DD/POWER = ON
03:37:55 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:55 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}
03:37:56 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:56 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":47}
03:37:57 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:57 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":49}
03:37:57 RUL: SWITCH1#STATE=4 performs "cmnd/tasmota_0042DD/DIMMER +"
03:37:57 MQT: stat/tasmota_0042DD/RESULT = {"POWER":"ON","Dimmer":51}
03:37:58 RUL: SWITCH1#STATE=6 performs "mem1 +"
03:37:58 MQT: stat/tasmota_0042DD/RESULT = {"Mem1":"+"}

On 1/4/2021 7:07 PM, BPSoft wrote:
> Rule1
> ON System#boot mem1 + ENDON
> ON switch1#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON
> ON switch1#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON
> ON switch1#state=5 DO mem1 - ENDON
> ON switch1#state=6 [...]

BPSoft, 2021-01-05:
Maybe for Rule2 we have to create a different Variable, try this:

 Rule1
 ON System#boot mem1 + ENDON
 ON switch1#state=2 DO cmnd/tasmota_0042DD/POWER TOGGLE ENDON
 ON switch1#state=4 DO cmnd/tasmota_0042DD/DIMMER %mem1% ENDON
 ON switch1#state=5 DO mem1 - ENDON
 ON switch1#state=6 DO mem1 + ENDON

 Rule1 1

 Rule2
 ON System#boot mem2 + ENDON
 ON switch2#state=2 DO cmnd/tasmota_0042DD/POWER TOGGLE ENDON
 ON switch2#state=4 DO cmnd/tasmota_0042DD/DIMMER %mem2% ENDON
 ON switch2#state=5 DO mem2 - ENDON
 ON switch2#state=6 DO mem2 + ENDON

 Rule2 1

For the double tap feature i've mixed feelings, because Tasmota Docs says:

> SwitchMode 12 Set switch to inverted pushbutton with dimmer mode **incl. double press feature.**

And internet says that it's only possible with Button mode.
Doc regarding that route:
https://tasmota.github.io/docs/Rules/#button-single-press-double-press-and-hold

Apparently Switch<x>#state=8 is for doubletap, we could try something like:

 Rule1
 ON System#boot mem1 + ENDON
 ON switch1#state=2 DO cmnd/tasmota_0042DD/POWER TOGGLE ENDON
 ON switch1#state=4 DO cmnd/tasmota_0042DD/DIMMER %mem1% ENDON
 ON switch1#state=5 DO mem1 - ENDON
 ON switch1#state=6 DO mem1 + ENDON
 ON switch1#state=8 DO cmnd/tasmota_0042DD/DIMMER 100 ENDON

 Rule1 1

 Rule2
 ON System#boot mem2 + ENDON
 ON switch2#state=2 DO cmnd/tasmota_0042DD/POWER TOGGLE ENDON
 ON switch2#state=4 DO cmnd/tasmota_0042DD/DIMMER %mem2% ENDON
 ON switch2#state=5 DO mem2 - ENDON
 ON switch2#state=6 DO mem2 + ENDON
 ON switch1#state=8 DO cmnd/tasmota_0042DD/DIMMER 100 ENDON

 Rule2 1

rkhanso, 2021-01-05:
Thanks for those rules. It's pretty much working how I want to now. I 
just need to mess around with the switches and get used to how to change 
from dim or brighten them, especially when they're at 100% to start with.

I really do appreciate the help. And never would have been able to 
figure this out on my own since I'm new to Tasmota/Shelly/Home 
Automation in general.

I'd like to send you some money or something for helping. It appears 
you're not in the US (maybe Sweden?), so I'm not sure the best way to do 
that. Do you have PayPal?

Have a great day and Happy New Year!!

On 1/4/2021 11:48 PM, BPSoft wrote:
> Rule1
> ON System#boot mem1 + ENDON
> ON switch1#state=2 DO cmnd/tasmota_0042DD/POWER TOGGLE ENDON
> ON switch1#state=4 DO cmnd/tasmota_0042DD/DIMMER %mem1% ENDON
> ON switch1#state=5 DO mem1 - ENDON
> ON switch1#state=6 DO mem1 + ENDON
> ON switch1#state=8 DO cmnd/tasmota_0042DD/DIMMER 100 ENDON