jamesturton/shelly-dimmer-stm32 #16: Rotary Encoder working with Shelly Dimmer 2

BPSoft, 2021-01-03:
Probably opening as an issue here is not the best option, but as the tasmota shelly dimmer community should be here i think i just have to share. 

I've a rotary encoder (PEC11L without On/Off switch) connected to a Shelly DImmmer 2, ground of shelly pinheader to encoder center PinC, SW1 to PinA, SW2 to PinB.
Conection is similar to Instructable: 
[https://www.instructables.com/Shelly-Dimmer-Wall-Switch-With-Rotary-Knob-and-Hom/](url)

In Tasmota library "support_rotary.ino",  lines 139 and 140 we have:
      pinMode(Encoder[index].pina, INPUT_PULLUP);
      pinMode(Encoder[index].pinb, INPUT_PULLUP);

Our encoder is conected as active to low (GND), for security reasons (no AC isolation), and Shelly hardware converts ground signal as logic 1, so we can't have an INPUT_PULLUP.

I've changed to:
      pinMode(Encoder[index].pina, INPUT);
      pinMode(Encoder[index].pinb, INPUT);

Compiled with "user_config_override.h" as:
#define ROTARY_V1                                         // necessary include
#define ROTARY_MAX_STEPS     100               // Rotary step boundary, default was 10

Resulting Tasmota binary:
[tasmota _dimmer.bin.gz](https://github.com/jamesturton/shelly-dimmer-stm32/files/5761912/tasmota._dimmer.bin.gz)

Changed your Template as image:
[image]

Create a Rule:
Rule1
   ON Rotary1#Pos1 DO Dimmer %value% ENDON
Rule1 1

And the mandatory SetOption:
SetOption98 1

//My actual SetOptions
Backlog LedTable 0; dimmerstep 5; DimmerRange 15,100; SwitchMode1 11; SwitchMode2 11; SwitchDebounce 40; SetOption32 1; SetOption13 1; SetOption99 1; SetOption93 1; SetOption40 1; SetOption34 1; SetOption1 1, SetOption98 1

For Shelly Dimmer 1, change the template accordingly:
switch_n<1> to rotary_a<1>
switch_n<2> to rotary_b<1>

If rotation is inverted, just swap  rotary_a with rotary_b

Everything is working extremely well, the change i've done will impact functionality on other devices like Module "Mi Desk Lamp (66)", so a Tasmota pull request is out of question for now.

@jamesturton maybe a future Shelly Module could ease this as the already similiar code in support_rotary.ino:
      #ifdef ESP8266
        if (MI_DESK_LAMP == TasmotaGlobal.module_type) {
          Rotary.model = 0;
        }
      #endif  // ESP8266

maybe is possible to associate pinMode per device.

jamesturton, 2021-01-04:
Hi @BPSoft! Great to hear that you are using this firmware :) Seems like a good setup you have using the rotary switch! If you copy this issue over to the Tasmota repo I will modify the Shelly dimmer driver to set the rotary switch options correctly when using this module.

BPSoft, 2021-01-05:
@Jason2866 
Issue created at https://github.com/arendst/Tasmota/issues/10407

Shelly Dimmer was installed on a floor lamp where an older Triac Dimmer was buzzing after a LED conversion.
So a rotary encoder was almost mandatory, visually a 1:1 conversion.

Allterco included a Rotary Switch on his firmware 1.8.0 and removed it later without correct operation of the switch.

It's funny to see this things!!!

Maikel-K, 2021-01-08:
This looks like something I want to try to.
I now have a KY-040 rotary encoder connected to the programming headers like this:
[image]
[image]

I know it looks a bit dodgy but it work fine now. I installed ESPEasy on the Dimmer2 and use it as a remote for another Dimmer2 via HTML requests.

I’m also using the push on/off function. 

It would be great if we can use up/down and on/off function. 

If I flash your posted bin file it should work with the dimming at least?

Jason2866, 2021-01-08:
@Maikel-K  Shelly dimmer 2 with this stm32 and Tasmota firmware does fully support a rotary encoder. Use latest Tasmota development version

BPSoft, 2021-01-12:
@Maikel-K it's working fine

You only need 2 wires from the Shelly pinheader, GND and GPIO0, forget 3.3V, not necessary and unsafe.

If you use wire with copper of 0.5mm diameter, you could connect directly on the pinheader

If you use KY-040 connect like this:

CLK to SW1 or SW2
DT to SW2 or SW1
SW to GPIO0
(+) to nothing
GND to GND

[image]

Flash with the latest development binaries for Tasmota firmware, they are [here](http://ota.tasmota.com/tasmota/)
Choose tasmota.bin.gz for OTA/gzipped, or tasmota.bin for uncompressed serial flash.

Then use this Template:
`{"NAME":"Shelly Dimmer 2 (Rotary)","GPIO":[32,3200,0,3232,5568,5600,0,0,6272,0,6304,0,0,4736],"FLAG":0,"BASE":66}`

jamesturton, 2021-01-12:
Great work @BPSoft with getting this working!

closed 2021-01-12 by jamesturton

OleElgaard, 2021-01-20:
I would love to be able to dim the light with a rotary! Do you have any more photos, or video? How should it be programmed?

BPSoft, 2021-01-20:
@OleElgaard
Shelly Dimmer 1 or 2?
Are you still with original Shelly firmware, or Tasmota?

OleElgaard, 2021-01-20:
dimmer 2. I only know of shelly firmware.  would love to learn tasmota :)

BPSoft, 2021-01-20:
My advice is to start converting to Tasmota:

[mgos-to-tasmota](https://github.com/yaourdt/mgos-to-tasmota)

Read that page and see the video tutorial.

All the flashes are OTA, Over The Air.
I've never encounter any trouble, but is advisable to be prepared for serial/physical flash if something goes wrong.

If you are prepared, or ready to assume minimum risk, go for it.

After you have Tasmota in your device, you should flash firmware for the STM32 co-processor on the Shelly Dimmer 2.

https://github.com/jamesturton/shelly-dimmer-stm32

Your first objective should be running Tasmota with your standard switches, like you already have with Shelly Firmware.

Only then we could think about the rotary encoder process. Do you already have the rotary encoder with you?

Please post your rotary model or a photo, and i will assist you.

trvrnrth, 2021-03-09:
@BPSoft I'm using a KY-040 rotary encoder with switch along with ESPHome. This is all working fine with CLK/DT hooked up to SW1/SW2 and SW to GPIO0 except that I need to activate the on-board (external) pullups in order to avoid a phantom rotation on switch press. I'm curious how you'd avoid this? I guess I could use GPIO1/GPIO3 (TX/RX) with the internal pullups activated instead of SW1/SW2 which would presumably be preferable to taking the 3.3v for the external pullups.

no1knows, 2022-01-22:
Any reason Tasmota might fry my Dimmer 2? I wired it up as shown - worked fine with a dimmable LED bulb in testing. Wired it up in my wall and the (5 x 10W dimmable LED) lights were flickering a lot, making a buzzing noise, and the Dimmer 2 ADC temperature was reaching ~95C and cutting out. I then tried ShdLeadingEdge 1 (i.e. trailing edge) to see if that would help, and the Dimmer 2 died as soon as I toggled it. I can hear a faint "buzz" "buzz" "buzz" when it has power, which suggests to me that it's in some kind of bootloop (or otherwise unable to power up). I managed to break some of the pins so can't attach UART.

BPSoft, 2022-01-23:
@no1knows mine is still working and going strong! I'm not using the last STM32 coprocessor firmware release, I've tried but gave me severe unstable flickering problems, and reverted back. In practice is not easy to implement an universal led dimmer, because we need to detect ac frequency, do calculation, zero crossing resync, leading/trailing switch, etc... Then we still have the power factor of the LEDs giving some problems. I must remember that Shelly original firmware also fry on some setups.

jamesturton, 2022-01-24:
@no1knows Take a look at this issue. https://github.com/jamesturton/shelly-dimmer-stm32/issues/10

Without any more information it is hard to tell exactly, but it sounds like a very similar to the hardware issue Roxxor91 had. If so, you can contact Shelly directly and try to get a replacement unit under warranty. It seems like this hardware has a very high failure rate and very few/no safety features in the hardware. In this case, it is safe to say "you get what you pay for".

no1knows, 2022-01-25:
Thanks very much. There are no visible "letting smoke out" signs on the board, but I managed to solder connections onto the UART pins and there's no output at all (and the ESP8266 quickly starts to get very warm...).

On the next attempt perhaps I'll try out the original STM32 firmware release.

frank8the9tank, 2022-03-17:
Has anybody got it to work trough the pin header for the rotary encoder? I want to control the shelly 2 only trought the pin header, which leaves the 2 normal input (the blue connect) free to use in a normal "hotel" switching arengement.

sam150122, 2022-07-16:
Is there another place I can pick up the ground and gpio 0, I've broken my header off the board? Need to find somewhere to solder the wires to?