jamesturton/shelly-dimmer-stm32 #17: Flash when switched on and during reboot

stefxx, 2021-01-04:
Hi James! Thanks again for your great work. I am using it currently with 6 Dimmer 2 modules, all running 9.2.0 firmware. However, I have a minor issue with all of them...

When I switch on the Dimmer 2, I often (but not always) see a short "flash" (full brightness) before the light goes to the last brightness level. It happens when using the local switch, but also by using and MQTT. Also, when the module reboots, I see the same thing happening. Even when the light is off (so it was off before the reboot, if flashes very shortly and than remains off).

Not the biggest issue in the world but if you see an opportunity to resolve this, that would be much appreciated. Thanks!

jamesturton, 2021-01-04:
Thanks for testing! Will take a look and try and fix this :)

jamesturton, 2021-01-04:
Can you try this version and let me know if it helps?

[shelly-dimmer-stm32_v51.5_beta-1.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/5765554/shelly-dimmer-stm32_v51.5_beta-1.zip)

stefxx, 2021-01-04:
Thanks! I installed the beta on one of my Dimmer 2 modules. There is a dimmable LED bulb connected to this dimmer.

It took me 5 reboots before it happened again. And I switch it on and off for about 10 times, but then I was able to reproduce the same error. It feels like it is a bit less than before, but certainly not gone. And since it happens only every now and than it is hard to say if it is really better or not.