jamesturton/shelly-dimmer-stm32 #18: Shelly Dimmer 2 - Voltage Monitor

BPSoft, 2021-01-22:
I've no info about what wrong values we should get with unmodified Shelly Dimmer 1, but It seems voltage monitoring is working on Shelly Dimmer 2 without hardware change.

Do you think it's possible/usefull to turn on SHELLY_VOLTAGE_MON by default, or by template separation detection?

[image]

A power factor of 1.00 with a Led Lamp is to good to be true, and regarding Total Power, let's just cross my fingers. 😅

jamesturton, 2021-02-20:
The Shelly dimmer 2 does indeed have working power (voltage and current) monitoring, but it's a little bit tricky to calculate the phase angle for the power factor. Definitely something I can look into getting working. It would be nice to have.
I'll see if I can make the changes in Tasmota to show the power setting only if it's being used with a dimmer 2 as I don't want to show it for dimmer 1 (if someone is eager enough to move a resistor on their device, they should be capable of building Tasmota with the SHELLY_VOLTAGE_MON flag... maybe?)

jamesturton, 2021-05-02:
Pull request has been made which fixes this! https://github.com/arendst/Tasmota/pull/11988

closed 2021-05-02 by jamesturton