jamesturton/shelly-dimmer-stm32 #22: Hue Emulation Problem -> Alexa wont control

masterpwarz, 2021-04-02:
Hey,
does anybody also have Problem with the Hue emulation of this firmware?
Alexa can discover it but cant control it. Wemo works fine but its a Dimmer so WeMo is useless.

log is always the same 

01:08:09 UDP: Packet (119)
01:08:10 HTP: Hue-API () from 192.168.178.130
01:08:10 HTP: Hue POST-Argumente ({"devicetype": "Echo"})
01:08:10 HTP: Hue Authentication Result ([{"success":{"username":"ad3916"}}])
01:08:10 HTP: Hue-API (/ad3916/lights) from 192.168.178.130
01:08:10 HTP: Hue Result ({"181637473":{"state":{"on":true,"bri":204,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Couchlicht","modelid":"Shelly Dimmer ","manufacturername":"Tasmota","uniqueid":"84:cc:a8:ad:39:16:00:11-1"}})
01:08:10 UDP: Packet (119)

Does someone has this Problem or ideas?

daxy01, 2021-04-02:
I'm afraid you're in the wrong place.
This is the low-level driver for Shelly dimmers which only takes care of the dimming piece in Tasmota.
I am pretty sure you should open an issue at the Tasmota branch which takes care of the higher-level things like Hue emulation.

masterpwarz, 2021-04-02:
Ah, okay
I assumed this meight be a driver problem, because the origin Tasmota 9.1 works fine with hue emulation, just this fork didn't. But I meight be wrong with that. 
Sorry for that.

closed 2021-04-04 by jamesturton