jamesturton/shelly-dimmer-stm32 #25: Feature Proposal: Detect short interruption of mains

rnauber, 2021-04-29:
Hi @jamesturton,

I like to retrofit some mechanical switches to become "smart" with the Shelly Dimmer 2. However, as I cannot fit them inside the switch box, I came up with this idea:

[image]

The retrofit adds the Shelly to the lamp (where there is ample space in the lampshade). In the switch box, a simple wire bridge is added. This means that the Shelly is powered all the time, except for a brief period during switching. 

I like to ask, if you could add a mechanism for detecting these brief interruptions to the firmware? I have not investigated their precise length, but they seem quite short, as they do not reset Shelly's ESP.

What do you think?

Best,
Richard

rbswift, 2021-04-29:
I do something similar but have replaced the rocker switches with momentary, normally closed switches. PowerOnState 2 is probably what you are looking for. It's a core Tasmota feature, not specific to the dimmer firmware. You'll need to be sure power is removed for long enough for the device to power down of course. Booting is quite fast so that's good. Also look out for SetOption36 and SetOption65.

rnauber, 2021-04-30:
> I do something similar but have replaced the rocker switches with momentary, normally closed switches. PowerOnState 2 is probably what you are looking for. It's a core Tasmota feature, not specific [...]

And that is exactly the problem, as I want to use the very brief interruption that occurs when switching a toggle  switch. As I wrote earlier, it does not reset/reboot the ESP.

jamesturton, 2021-05-02:
This is an interesting proposal and I can see the benefit it would have to your setup.
If this was implemented I wonder about the reliability of the detection. I think some good first steps would be:
- Determine exactly how long it takes for the ESP to restart after being disconnected.
- Determine how long of a 'pulse' is generated from a switch in this configuration (test multiple brands of switch to see if it is consistent)
- See if the pulse of this length is detectable by the STM

Jason2866, 2021-05-02:
If the ESP reboots it will take a few seconds until it is online again.
Perfect would be if the interrupt is very short so that the ESP does not reboot.
Needs testing...
If this is the case STM should detect (if possible) and do the magic.
Intersting in real world will be if and how many ghost switching will happen

rbswift, 2021-05-02:
The delay to get online typically doesn’t matter much. Toggling the light on/off is sub-second fast.

A problem with a very short interruption is that it could easily be falsely triggered by switching in the supply network.

If you can replace your switch with a momentary push button, normally closed switch mechanism then poweronstate toggle works brilliantly and also has the advantage that the switch doesn’t sometimes appear to be ‘on’ when the light is actually switched off due to automation/ remote control. Did you try this or is it not possible to replace your switch?

rnauber, 2021-05-02:
> The delay to get online typically doesn’t matter much. Toggling the light on/off is sub-second fast.
> 
> A problem with a very short interruption is that it could easily be falsely triggered by switching in the supply network.

Yes, but I hope that is rare enough to be tolerated.

> If you can replace your switch with a momentary push button, normally closed switch mechanism then poweronstate toggle works brilliantly and also has the advantage that the switch doesn’t sometimes [...]

My particular switches can not be modified that easy. 
But in general, if you can have an *normally closed momentary switch* that would be the best option.