jamesturton/shelly-dimmer-stm32 #26: Light too 'dim'

LaborEtArs, 2021-11-10:
Just installed Tasmota to use it with the Shelly Dimmer 2
Hopefully I got the most recent versions for Tasmota and for the STM32 co-processor
The console shows:
```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 81
00:00:00.056 QPC: Count 1
00:00:00.066 Project tasmota - Tasmota Version 10.0.0(tasmota)-2_7_4_9(2021-10-19T07:53:37)
00:00:00.067 SNS: Hardware Serial
00:00:00.171 SHD: Shelly Dimmer Co-processor Version v51.6
00:00:00.565 WIF: Connecting to AP1 AirportT8 Channel 8 BSSId 2E:80:88:DC:A4:97 in mode 11n as tasmota-19E228-0552...
00:00:04.753 WIF: Connected
00:00:05.005 HTP: Web server active on tasmota-19E228-0552 with IP address 192.168.1.41
15:02:46.013 RSL: INFO1 = {"Info1":{"Module":"Shelly Dimmer 2","Version":"10.0.0(tasmota)","FallbackTopic":"cmnd/DVES_19E228_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:02:46.016 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-19E228-0552","IPAddress":"192.168.1.41"}}
15:02:46.017 RSL: INFO3 = {"Info3":{"RestartReason":"External System"}}
15:02:46.019 RSL: RESULT = {"POWER":"OFF"}
15:02:46.020 RSL: POWER = OFF
15:02:46.466 QPC: Reset
15:02:50.405 RSL: STATE = {"Time":"2021-11-10T15:02:50","Uptime":"0T00:00:12","UptimeSec":12,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AirportT8","BSSId":"2E:80:88:DC:A4:97","Channel":8,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:02:50.414 RSL: SENSOR = {"Time":"2021-11-10T15:02:50","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":43.1},"ENERGY":{"TotalStartTime":"2021-11-07T21:09:20","Total":0.002,"Yesterday":0.000,"Today":0.002,"Period": 0,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":1.00,"Voltage":232,"Current":0.002},"TempUnit":"C"}
```
For testing I used a 60W standard bulb.
When turning the light on (while the dimmer is set to 100%), the bulb is too 'dim'. This optical impression is backed up by the power reading, which shows just 36W (for a 60W bulb).
In addition to this, the bulb flickers.
Is there something I can do against this behavior? Am I asking the question on the right place, or should I create an issue directly in the Tasmota repository?
Many thanks for any help in forward!

LaborEtArs, 2021-11-13:
Just found, that setting 'DimmerRange 50 200' in the Tasmota console 'solved' the problem. Obviously something with the calibration is faulty. I did not run the original Shelly firmware in front of installing Tasmota. Should I have done so?

jamesturton, 2021-11-15:
Hi @LaborEtArs! I'm glad you managed to fix your issue. Did you also change the `Ledtable 0` option to disable the Gamma correction like it describes on the [Tasmota Device Repository](https://templates.blakadder.com/shelly_Dimmer_2.html)?

closed 2021-11-15 by jamesturton

LaborEtArs, 2021-11-18:
Jep, I followed exactly these instructions :-) They led me to the 'DimmerRange' experiment, which was finally successful!