jamesturton/shelly-dimmer-stm32 #27: Light blinks with full brightness instead of dimming on Dimmer 2 on release 51.6

Leo-PL, 2021-12-12:
When flashed with Tasomota 10.0.0.0, and STM32 firmware release 51.6, the module exhibits strange behaviour, just as if the zero-cross detection wasn't operational. Light is blinking with a very low frequency, around 1Hz, and duty cycle somewhat proportional to the brightness set in Tasmota. When attempting to turn off the light in that state, it remains fully-on until Tasmota is restarted.

With firmware 51.5, dimming actually works properly.

Behaviour is the same, regardless if `ShdLeadingEdge` is set or cleared.

My test setup is a 100W halogen bulb, connected with neutral wire on Dimmer 2. On stock firmware, the dimming works properly.
I wonder if there isn't an even newer hardware revision, so attaching photos of my unit:
[image]

[image]

jamesturton, 2021-12-13:
@Leo-PL Thanks for the pictures! I will check my old Dimmer 2 to see if anything looks different. If it does look different, I might ask you to make some tests if that would be ok?

As a little bit of background - the zero crossing detection is done in this firmware. Normally that kind of thing would be done in hardware, but of course, adding additional circuitry would increase the cost of the device, whereas doing it in firmware is 'free'.

tzieg-2021, 2021-12-13:
**@jamesturton**, I can confirm Leo-PLs observation:
- With Tasomota 10.1.0, and STM32 firmware release 51.6, I also see blinking, strange dimmer ranges, not reaching the 100%, etc. for both leading and trailing edge, even for a regular bulb, i.e. no LED.
- Using 51.5 beta-2 solved all issues.
Seems not everything made it from 51.5-beta-2 into 51.6.

Regarding the diffent flavors of the Shelly Dimmer 2: There is a [thread on the shelly-support.eu forum](https://www.shelly-support.eu/forum/index.php?thread/4782-shelly-dimmer-2-firmware/&postID=129145#post129145) with pictures of two different versions.
-  The one Leo-PL did show in his message above
-  One with "2 black wires" (from a new diode I think) seems to be a newer V2-version

I blew up my V2-version most likely by accidentally driving halogen LEDs via a transformer while having the trailing edge activated. In [thread #30](https://www.shelly-support.eu/forum/index.php?thread/4782-shelly-dimmer-2-firmware/&postID=130075#post130075), Thomas Goebel did a great analysis on my damaged Dimmer-2, he also created a circuit diagram of the vertical daughter board.

Leo-PL, 2021-12-13:
@jamesturton sure, I'll be happy to test.

BusinessClaes, 2022-01-02:
Hi!
I get the same behaviour, both with Tasmota 10.0.0 and 10.1.0
But when I do a ShdLeadingEdge 1 and then back to 0 the problem disappears.
Hope this helps finding the bug.
Cheers

Leo-PL, 2022-01-03:
I tried the same, but in my case (purely resistive load) this didn't change a thing.
I'm not using this dimmer "in production" yet - it was given to me by a friend - so I can test it on the bench with current limit, if needed.

poldim, 2022-07-30:
I'm experiencing the same thing.  I thought it was an issue with ESPHome so I made an issue there but later tried tasmota and saw the same behavior.  Tried downgrading to 51.5 on a whim and it's working just fine.

I tried to time the blinks and they appear to be about .66 seconds on my 60Hz light.

@jamesturton any thoughts on what this could be?

no1knows, 2022-09-25:
+1 also seeing this behaviour (or no light at all) on Tasmota 11.0.0 and 51.6. All working find on 51.5 (although with a lot of flickering at low brightness).