jamesturton/shelly-dimmer-stm32 #29: Overcurrent protection

edge90, 2022-03-28:
Would be cool if the firmware shut the output off if overcurrent is detected, like the standard shelly does.

jamesturton, 2022-03-28:
Should be possible! Do you have any suggests on how it could be implemented?
I was thinking: if the unit detects the over-current it turns off the output and then the unit must be restarted (disconnected and connected to mains voltage) to get it working again. That way it forces the user to do something and hopefully they will check the unit is ok before it can be used again.
Do you know if ESPHome/Tasmota support notifications for over-current events? It would be nice if an error message could be displayed in the UI, rather than just not being able to turn on the output again.
The STM also has an in-built temperature sensor so over-temperature events could also be detected too.

edge90, 2022-03-28:
Not sure about tasmota but esphome could just expose a sensor or so (that would be visibile in home assistant for those running that). Exactly what I was thinking, disable the interups and then call `mosfet_off()`. As of now, the current is measured after receiving the poll command from the esp. This is bad from a safety pov as it requires the esp to initiate the process.

tobby88, 2022-04-13:
> ...then the unit must be restarted (disconnected and connected to mains voltage) to get it working again.

A reboot would be a good idea - but to reconnect would be bad, since most of these devices are mounted into the wall.

oliv3r, 2025-01-10:
In esphome it seems to be common to do this:

```yaml
  - platform: shelly_dimmer
    current:
      on_value_range:
        - above: 70
          then:
           - light.turn_off: dimmer
```