jamesturton/shelly-dimmer-stm32 #30: Shelly Dimmer 2 occasionally becomes unavailable

ClioOpen, 2022-07-02:
Hello, I have recently flashed two Shelly Dimmer 2 devices following the instructions available on [Blakadder](https://templates.blakadder.com/shelly_Dimmer_2.html). I notice that both devices sporadically become unavailable in the Home Assistant and I need to flip the fuse off and back on again to get it back "online".

Anyone experienced anything similar?

ClioOpen, 2022-07-02:
Adding just in case if anyone stumbles here while searching through the web. Feel free to skip the paragraph below as I'm throwing in as much of search phrases as I can; go straight to the last one.

My Tasmota flashed device disconnects from the WiFi. Shelly with Tasmota device regularly becomes unresponsive until I restart it.  Shelly module unavailable. Flashed Shelly module unavailable on WiFi.  Shelly Dimmer 2 unavailable in Home Assistant. Shelly device disconnects from WiFi. Shelly tasmota connectivity issues with Asus router.

This is, as it appears, a generic IoT topic. at least for me. I use an Asus router and had Smart Connect enabled; it works great with my other devices and the experience is seamless. The switches dislike it though; [Asus recomendation](https://www.asus.com/us/support/FAQ/1042475). Surprisingly, original Shelly firmware apparently handled it.

jamesturton, 2022-07-02:
Hey @ClioOpen!

I would suggest that you post this issue on the Tasmota repo as this repo is only regarding the firmware flashed to the stm32. I believe that it is now possible to use ESPHome with the Shelly Dimmer (although I have not tried it myself) so if you need the 'Smart Connect' enabled and don't manage to get the issue solved with Tasmota you might be able to try with that.

Hope that helps!

ClioOpen, 2022-07-03:
Thanks! Valid points.

closed 2022-07-10 by jamesturton