jamesturton/shelly-dimmer-stm32 #31: only 50% brightness on dimmer2

smennie, 2022-07-20:
Hi there,
I've noticed that the 51.6 code has max_brightness set to 500 for dimmer2 and 1000 for dimmer1.
For my dimmer2 this results in ~50% max brightness.
When changing max_brightness to 1000, I get full brightness, however a mixture of very odd and unpredictable behavior everything from flickering, to the light being stuck on.

Is there any reason why max_brightness is set to 500?  On my LED down light which supports both leading and trailing edge, the brightness is obviously very low when max_brightness is set to 500. 

Going back to 51.5, the brightness issue is resolved, however getting flickering issues which I understand have been reported before and largely resolved in 51.6

jamesturton, 2023-09-12:
Hi @smennie!
Max brightness was capped to 50% to get it working with no-neutral setups. In the latest release I have now removed the no-neutral code so I hope that it fixes the problems you were having!
https://github.com/jamesturton/shelly-dimmer-stm32/releases/tag/v51.7

smennie, 2023-09-16:
Thank you sir! I've thrown v51.7 on and it appears to be working flawlessly.

jamesturton, 2023-09-16:
Great to hear it's working well!

closed 2023-09-16 by jamesturton