jamesturton/shelly-dimmer-stm32 #33: Just flickering no dimming or turn off function with stm32_v51.6

TKaluza, 2023-02-25:
Most things work without a problem in 51.5 but with 51.6 my shelly dimmer 2 are not working in an way. it seems to be flickering dimmed to 50% with no option to switch it off. Any ideads?

jamesturton, 2023-02-25:
Hi @TKaluza!
What wasn't working for you in 51.5? Why did you decide to upgrade?
Are you using a neutral wire in your setup?

gwinkless, 2023-07-11:
Same here. While the device is currently plugged in to a socket with a neutral, the whole reason I got the Shelly in the first place was to move location to in-switch (with a rotary for the dimmer), and that definitely won't work since we're in the UK (traditionally no neutrals in-switch here).

Is the neutral wire actually problematic for v51.6? Should I disconnect it from the Shelly and try it without?

shuntfield, 2023-09-11:
I have just bought two new shelly dimmer 2 - and they are exhibiting this behaviour with the 51.6 with the 51.5 they are fine. I am finding if you have the neutral on the dimmer, then the flicker isn't as intense as it is without the dimmer. I also have an older dimmer 2 which works fine, I think there may be a hardware change on the new one. 
I have tried it with an Incandescent lamp and also a dimmable LED and the behavour is similar.

shuntfield, 2023-09-11:
Yup @jamesturton I can I have an "older" one here as well which I can compare.

Will post later today or tomorrow.

shuntfield, 2023-09-11:
@jamesturton I don't know if it's possible I'm not that great with the code, but would it be possible as an interim fix to get the 51.5 and add the SHDwarmupbrightness section to it? Is that possible?

jamesturton, 2023-09-11:
> @jamesturton I don't know if it's possible I'm not that great with the code, but would it be possible as an interim fix to get the 51.5 and add the SHDwarmupbrightness section to it? Would that work?

I'll try and get you a version for you to test tonight!

shuntfield, 2023-09-11:
@jamesturton awesome.
It's all in bench testing so I can play about with it in the lab!
Fantastic!

jamesturton, 2023-09-11:
I have made a branch based on the v51.5 tag with the warmup brightness command [here](https://github.com/jamesturton/shelly-dimmer-stm32/commits/shuntfield-test). This is the binary for you to test here: [shelly-dimmer-stm32-9241b1a.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12580789/shelly-dimmer-stm32-9241b1a.bin.zip)

I would also like if you could test this version for flickering which is based of the main repository, just with the last couple of commits reverted: [shelly-dimmer-stm32-f1c1e7c.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12580800/shelly-dimmer-stm32-f1c1e7c.bin.zip)

shuntfield, 2023-09-11:
OK I will download and install both.
I have the luxury of having two identical new dimmers.

An 'old' dimmer all of which I can bench.

I will install the 51.5 modified into one of the new, then the modified branch into the other of the new, then test both on the old dimmer.

When I have completed I will come back and let you know, it's midnight UK here so I will lab-it-out tomorrow and get back.

jamesturton, 2023-09-11:
Awesome! Thanks so much for your help!

shuntfield, 2023-09-11:
> Awesome! Thanks so much for your help!

No, absolutely thank you! If you can get the software to work it's amazing and will help me no end.

Will feed back asap.

shuntfield, 2023-09-12:
@jamesturton I'm trying the method to install the test binary and the message coming back from Tasmota "Invalid file signiture" I'm not sure if Tasmota is trying to update itself?

jamesturton, 2023-09-12:
> @jamesturton I'm trying the method to install the test binary and the message coming back from Tasmota "Invalid file signiture" I'm not sure if Tasmota is trying to update itself?
> If I reapply the 51.5 it works fine.

Ohh, strange... did you unzip the file first before trying to upload it? Unfortunately GitHub doesn't allow you to upload .bin files, so that's why I had to zip it.

If that doesn't work I'll take a look to see if I can see any changes in Tasmota which might have broken the integration.

shuntfield, 2023-09-12:
Yup I have unzipped it, perhaps there is something in the code for Tasmota that only allows the published Binaries to be loaded - I am going back to Tasmota 9.5 a moment to see if that works.

shuntfield, 2023-09-12:
I will open an issue on Tasmota, they may just tell me that I have to compile my own binary for it, I will see what happens!

shuntfield, 2023-09-12:
Meh - I was trying to apply the ZIP - it was unpacked in a different folder - ignore - now testing!

shuntfield, 2023-09-12:
OK Results are in:
"New" dimmer:

[shelly-dimmer-stm32-f1c1e7c.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12580800/shelly-dimmer-stm32-f1c1e7c.bin.zip)

Works fine no flickering

[shelly-dimmer-stm32-9241b1a.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12580789/shelly-dimmer-stm32-9241b1a.bin.zip)

Also works fine - with no flickering

Both cause flickering in the no-neutral setup. (which isn't a problem for me but tested for completeness)

Both work fine with Neutral on Incandecent and Dimmable LED

I just tried it on the "old" dimmer which I thought was working with the 51.6 but on original 51.6 it flickers, strangely, but with the new ( [shelly-dimmer-stm32-f1c1e7c.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12580800/shelly-dimmer-stm32-f1c1e7c.bin.zip) ) file it works fine

I'd say thats a success for this problem - SHDwarmupbrightness is working on f1c1e7c as well.

Thanks!

jamesturton, 2023-09-12:
Great to hear it's working better for you now @shuntfield!

I've pushed some changes to make the no-neutral code optional as I think that was what might have been causing the flickering problems you were having. I've got a new version if you wouldn't mind testing? [shelly-dimmer-stm32-5045243.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12587313/shelly-dimmer-stm32-5045243.bin.zip) If this version looks good to you in your setup with neutral connected then I will release an official version.

I was hoping that I could use one binary for both neutral and no-neutral setups, but if it causing flickering then I guess I will have to change to releasing two versions.

jamesturton, 2023-09-12:
> If that's a no neutral version I can test that too.

Here is a no neutral version for you to test too! 
[shelly-dimmer-stm32-5045243-no-neutral.bin.zip](https://github.com/jamesturton/shelly-dimmer-stm32/files/12587793/shelly-dimmer-stm32-5045243-no-neutral.bin.zip)

closed 2023-09-12 by jamesturton

shuntfield, 2023-09-12:
Awesome - I will install the release now - Thanks Great Job!