jamesturton/shelly-dimmer-stm32 #35: Reduced brightness compared to Shelly firmware

DerDom, 2023-09-12:
Hey, I recognized that my LED lamps aren't as bright after flashing Tasmota and this STM32 firmware. Besides, Tasmota shows the power consumption to be at 26W, while the manufacturer claims a power consumption of 30.4W. 

Additional information:
- I am using 8 of these GU10 LED bulbs: https://www.amazon.de/dp/B073FT1QT2
- 8 x 3.8W = 30.4W
- Ledtable 0
- ShdLeadingEdge 0
- DimmerRange Min 14, Max 100

Unfortunately, I did not check the power consumption when using Shelly's original firmware. Please let me know if I can provide any additional information.

jamesturton, 2023-09-12:
Hi @DerDom! If you are using a neutral cable in your setup please can you try the version I've just released?
https://github.com/jamesturton/shelly-dimmer-stm32/releases/tag/v51.7

DerDom, 2023-09-13:
Wow, @jamesturton, thank you for your promt reply and for this quick release! I just upgraded Tasmota Version 13.1.0 from v51.6 to v51.7 and the lights got significantly brighter. They are now consuming 40W at full brightnes (according to shelly / tasmota). 

So I guess I am using 390lm/4.6W x 8 LEDs, not 345lm/3.8W x 8 LEDs (Philips produced different versions). And yes, I am using a neutral wire in my setup.

closed 2023-09-13 by DerDom

DerDom, 2023-09-13:
PS: I was also able to reduce DimmerRange to 6.