jamesturton/shelly-dimmer-stm32 #37: Template not working anymore

themac11, 2023-10-01:
Used Tasmota on my Shelly Dimmer 2 for months but now I have the problem that it is not recognized as a dimmer anymore.
The Template from blakadder
 {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18} uses 
"GPIO04 | SHD Boot 0 and GPIO05 | SHD Reset" with is not available in the dropdown menu in Tasmota.

Any advice?

jamesturton, 2023-10-02:
Hey @themac11!
You are best asking this question on the [Tasmota repo](https://github.com/arendst/Tasmota/issues/new/choose) as this repo only deals with the firmware on the STM32, not Tasmota running on the ESP8266.

closed 2023-10-02 by jamesturton

themac11, 2023-10-03:
Got help ion Tasmota Discord. Problem was the automatic update on TasmotaAdmin with the wrong binary for sensors. Had to use the normal bin.