jamesturton/shelly-dimmer-stm32 #38: Shelly Pro Dimmer 2PM

justspacedog, 2024-04-26:
Hi,

please add support for this device:
https://kb.shelly.cloud/knowledge-base/shelly-pro-dimmer-2pm

Thank you!

jamesturton, 2024-04-26:
Hi @justspacedog!

Please can you send a photo of how the PCB looks

justspacedog, 2024-04-28:
[image]
[image]

Can you work with those?

jamesturton, 2024-04-28:
Those photos are great, but they don't seem to show the STM chip. Can you see if you can find that on the board somewhere?

justspacedog, 2024-04-28:
[image]
[image]
[image]

Unfortunately I don’t really know what I’m searching for on the board. 🙈

CReimer, 2024-04-28:
You won't find a stm32 on this. The first picture clearly shows an EESP32-D0WDQ6
https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

Unfortunately this means Shelly completely changed their hardware. That's interesting, but out of scope of shelly-dimmer-stm32.

jamesturton, 2024-04-28:
Then you would be better asking in the Tasmota or ESPHome repo for them to add support their directly.

Out of curiosity, is it possible to read the markings on chip U11?

closed 2024-04-28 by jamesturton

oliv3r, 2025-01-05:
> Then you would be better asking in the Tasmota or ESPHome repo for them to add support their directly.

just out of curiosity, while I fully understand and agree that since there is no STM32, things are out of scope, what ARE they using?

I see a 595 shift-register to control the leds and maybe some IO.
Some chip SUPER close to the ESP32, which I can't read the labeling off.
The ade7953acpz is a single phase energy monitoring chip from analog.
U11 probably is the (SPI)PHY.

There seems to be some chips on the relay breakout boards as well, so it could be that that's where the STM32 lies. I would be surprised if they changed that part of their design as well. I get that some companies don't care much about re-use. The switch to ESP32 is probably related to more pins the 8266, desire to do LAN, but to find out a SPI PHY is easier to deal with then the MAC phy (also due to not enough pins maybe), or needing more 'pro' computing then the 8266 offers. But using the same serial protocol with the same stm32 still makes sense I'd reccon.

So @justspacedog could you also add some photo's of those relay boards?