jamesturton/shelly-dimmer-stm32 #41: Compatible with gen3 dimmer?

tbrasser, 2025-04-29:
Will probably get/flash one to find out. But would be great if people already tried it.

jamesturton, 2025-04-29:
According to [the website](https://www.shelly.com/blogs/documentation/shelly-dimmer-gen3) it's using a STM32g0 as a co-processor so at least some changes will be needed. If you do get one then please post photos of the PCB so I can begin reverse engineering to get it supported!

mikehosmar, 2025-05-20:
I can confirm that it uses a STM32g0.

[image]
[image]
[image]
[image]
[image]
[image]

dbrb2, 2025-09-19:
Did anyone manage to get this working, or is it still a WIP ?

splitice, 2026-02-22:
Unfortunately I don't have the gear or skills for that I think. I might need to accept defeat.

Beeping out traces isn't getting me anywhere. I'd need to disassemble the boards and work out the traces most likely.